ZhengyiLuo/PHC #4: Som tips for docker users

kexul, 2023-10-31:
Some tips for docker user: 

1. Download issaacgym and build a docker using its `build.sh` in its docker folder. 
Use base image `nvcr.io/nvidia/pytorch:23.02-py3` which uses last pytorch 1.x version. 
Add the gymuser to no-password sudoer by replacing `RUN useradd --create-home gymuser`.
```
RUN apt-get update && apt-get install -y sudo
RUN useradd --create-home gymuser
RUN adduser gymuser sudo
RUN echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers
```

2. Run docker container, note to change the DISPLAY enviroment variable to match your host. 
```
docker run -it --rm -v /tmp/.X11-unix:/tmp/.X11-unix -v "$HOME/.Xauthority:/home/gymuser/.Xauthority:rw" \
-e DISPLAY=localhost:11 --network=host --ipc=host --gpus=all   isaacgym /bin/bash
```

3. Download PHC code and install requirements by `pip install -r requirements.txt`. Here I replace the `sklearn` package with `scikit-learn` since sklearn is deprecated. 

4. Doanload SMPL data, rename and put them in the PHC folder's `data` folder, the file structure should be: 
```
PerpetualHumanoidControl/
├── assets
│   └── phc.gif
├── data
│   └── smpl
│       ├── SMPL_FEMALE.pkl
│       ├── SMPL_MALE.pkl
│       └── SMPL_NEUTRAL.pkl
```

5. Install mujoco: download `mujoco210-linux-x86_64.tar.gz`, unzip it and copy it to `/home/gymuser/.mujoco/mujoco210`
Update `LD_LIBRARY_PATH` enviroment variable. 
```
tar -xzvf mujoco210-linux-x86_64.tar.gz
mkdir /home/gymuser/.mujoco
cp -r mujoco210 /home/gymuser/.mujoco
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/gymuser/.mujoco/mujoco210/bin
echo 'export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/gymuser/.mujoco/mujoco210/bin' >> ~/.bashrc
source ~/.bashrc
```

6. Install `libosmesa6-dev` in case of `osmesa.h` error:
```
sudo apt install libosmesa6-dev
```
Now we should be able to run the test script: 
```
python uhc/smpllib/smpl_local_robot.py
python scripts/joint_monkey_smpl.py
```

7. Install gdown and download model files:
```
pip install gdown
sh downlaod_data.sh
```

8. Install `xephyr` to get `pyvirtualdesktop` working: 
```
sudo apt install xserver-xephyr
```
Now we should be able to run the evaluation script, viva!
```
python phc/run.py --task HumanoidImMCPGetup --cfg_env phc/data/cfg/phc_shape_mcp_iccv.yaml --cfg_train phc/data/cfg/train/rlg/im_mcp.yaml --motion_file sample_data/amass_isaac_standing_upright_slim.pkl --network_path output/phc_shape_mcp_iccv --test --num_envs 1 --epoch -1
````

ZhengyiLuo, 2023-10-31:
Also I really do need to remove the dependency on the old mujoco version...

ZhengyiLuo, 2023-10-31:
Updated to remove dependency on the old mujoco version.

kexul, 2023-11-01:
Glad that helps! 😄 will create a PR then.

closed 2023-11-01 by kexul