ZhengyiLuo/PHC #7: Evaluate on AMASS

kexul, 2023-11-10:
I tried to evaluate on amass using the command provided in README. 
```
python phc/run.py --task HumanoidImMCPGetup --cfg_env phc/data/cfg/phc_shape_mcp_iccv.yaml --cfg_train phc/data/cfg/train/rlg/im_mcp.yaml --motion_file sample_data/[amass_data].pkl --network_path output/phc_shape_mcp_iccv --test --num_envs 1024 --epoch -1 --im_eval
```

After modified the num_envs `1024` to `1`, and use motion file `sample_data/amass_isaac_standing_upright_slim.pkl`, the program could procced successfully in my 4090. 

But the three other sample_data load failed with the following errors: 
* sample_data/amass_copycat_occlusion_v3.pkl`
```
Traceback (most recent call last):
  File "phc/run.py", line 297, in <module>
    main()
  File "phc/run.py", line 291, in main
    runner.run(vargs)
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/torch_runner.py", line 142, in run
    player = self.create_player()
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/torch_runner.py", line 128, in create_player
    return self.player_factory.create(self.algo_name, config=self.config)
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/common/object_factory.py", line 15, in create
    return builder(**kwargs)
  File "phc/run.py", line 209, in <lambda>
    runner.player_factory.register_builder('im_amp', lambda **kwargs: im_amp_players.IMAMPPlayerContinuous(**kwargs))
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/learning/im_amp_players.py", line 28, in __init__
    super().__init__(config)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/learning/amp_players.py", line 23, in __init__
    super().__init__(config)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/learning/common_player.py", line 16, in __init__
    BasePlayer.__init__(self, config)
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/common/player.py", line 17, in __init__
    self.env = self.create_env()
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/common/player.py", line 136, in create_env
    return env_configurations.configurations[self.env_name]['env_creator'](**self.env_config)
  File "phc/run.py", line 193, in <lambda>
    env_configurations.register('rlgpu', {'env_creator': lambda **kwargs: create_rlgpu_env(**kwargs), 'vecenv_type': 'RLGPU'})
  File "phc/run.py", line 88, in create_rlgpu_env
    task, env = parse_task(args, cfg, cfg_train, sim_params)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/parse_task.py", line 60, in parse_task
    task = eval(args.task)(cfg=cfg, sim_params=sim_params, physics_engine=args.physics_engine, device_type=args.device, device_id=device_id, headless=args.headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im_mcp_getup.py", line 29, in __init__
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im_getup.py", line 59, in __init__
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im_mcp.py", line 22, in __init__
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im.py", line 88, in __init__
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_amp_task.py", line 37, in __init__
    super().__init__(cfg=cfg,
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_amp.py", line 109, in __init__
    self._load_motion(motion_file)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im.py", line 289, in _load_motion
    self._motion_train_lib = MotionLibSMPL(motion_file=motion_train_file, device=self.device, masterfoot_conifg=self._masterfoot_config, min_length=self._min_motion_len, im_eval=flags.im_eval)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_smpl.py", line 49, in __init__
    super().__init__(motion_file=motion_file, device=device, fix_height=fix_height, masterfoot_conifg=masterfoot_conifg, min_length=min_length, im_eval=im_eval, multi_thread=multi_thread)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_base.py", line 120, in __init__
    self.load_data(motion_file,  min_length = min_length, im_eval = im_eval)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_base.py", line 144, in load_data
    data_list = {item[0]: item[1] for item in sorted(self._motion_data_load.items(), key=lambda entry: len(entry[1]['pose_quat_global']), reverse=True)}
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_base.py", line 144, in <lambda>
    data_list = {item[0]: item[1] for item in sorted(self._motion_data_load.items(), key=lambda entry: len(entry[1]['pose_quat_global']), reverse=True)}
KeyError: 'pose_quat_global'
```
* sample_data/amass_isaac_gender_betas_unique.pkl
```
Traceback (most recent call last):
  File "phc/run.py", line 297, in <module>
    main()
  File "phc/run.py", line 291, in main
    runner.run(vargs)
[... 79 lines cut ...]
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im_getup.py", line 59, in __init__
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im_mcp.py", line 22, in __init__
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im.py", line 88, in __init__
    super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_amp_task.py", line 37, in __init__
    super().__init__(cfg=cfg,
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_amp.py", line 109, in __init__
    self._load_motion(motion_file)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/env/tasks/humanoid_im.py", line 289, in _load_motion
    self._motion_train_lib = MotionLibSMPL(motion_file=motion_train_file, device=self.device, masterfoot_conifg=self._masterfoot_config, min_length=self._min_motion_len, im_eval=flags.im_eval)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_smpl.py", line 49, in __init__
    super().__init__(motion_file=motion_file, device=device, fix_height=fix_height, masterfoot_conifg=masterfoot_conifg, min_length=min_length, im_eval=im_eval, multi_thread=multi_thread)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_base.py", line 120, in __init__
    self.load_data(motion_file,  min_length = min_length, im_eval = im_eval)
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_base.py", line 144, in load_data
    data_list = {item[0]: item[1] for item in sorted(self._motion_data_load.items(), key=lambda entry: len(entry[1]['pose_quat_global']), reverse=True)}
  File "/opt/isaacgym/motion_imitate/PerpetualHumanoidControl/phc/utils/motion_lib_base.py", line 144, in <lambda>
    data_list = {item[0]: item[1] for item in sorted(self._motion_data_load.items(), key=lambda entry: len(entry[1]['pose_quat_global']), reverse=True)}
IndexError: only integers, slices (`:`), ellipsis (`...`), numpy.newaxis (`None`) and integer or boolean arrays are valid indices
```

Many thanks!

ZhengyiLuo, 2023-11-10:
Hi @kexul, 

The other files in the sample data are supporting pkls that are used for data processing purposes:

- amass_copycat_occlusion_v3.pkl: for processing AMASS data, this file picks which files to remove due to human object interaction
- amass_isaac_gender_betas_unique.pkl: the unique body shapes from AMASS that does not trigger a issue in Isaac. Used to train body-shape aware body poses. 
- amass_isaac_gender_betas.pkl: all body shapes in AMASS. 

To test on the full AMASS, you will need to download and process the AMASS according to the (training instructions)[https://github.com/ZhengyiLuo/PerpetualHumanoidControl#training]. I will clear this up with a readme update.

closed 2023-11-10 by ZhengyiLuo

kexul, 2023-11-11:
Get it, many thanks for your explanation!