ZhengyiLuo/PHC #16: Failing in imitating mdm generated motions

Cr7st, 2023-12-06:
Hello! I've tried your model with some motion clips generated by mdm. However, the model is not doing well on theses clips. The following videos are some examples.

https://github.com/ZhengyiLuo/PerpetualHumanoidControl/assets/73263561/d4cd04d3-0506-47ea-9cc1-abf7ed36878d

https://github.com/ZhengyiLuo/PerpetualHumanoidControl/assets/73263561/be1cdda5-1b77-4adb-98ab-26d26c379e7e

Is it because of the low quality of the motion generated by mdm or other reasons? Do you have any suggestions to improve?
Thank you!

non-void, 2023-12-06:
https://github.com/ZhengyiLuo/PerpetualHumanoidControl/assets/19344848/a1a92361-a64b-46b3-93fe-eaec8896264c

https://github.com/ZhengyiLuo/PerpetualHumanoidControl/assets/19344848/fee24fc5-bef9-4071-b6e2-d9c5c85de569

ZhengyiLuo, 2023-12-06:
Two possible explanations:

- MDM generates motion assuming 20 fps motion? PHC assumes 30 for input, so the correct for needs to be used for the pkl file. The videos shown do seem the motion a little too fast.
- MDM's motion isn't really physically plausible, so some of the faster motion is expected to fail.

ZhengyiLuo, 2023-12-14:
Here is the code I used to convert mdm 20 fps to 30 fps, using interpretation. 
```
def fps_20_to_30(mdm_jts):
    jts = []
    N = mdm_jts.shape[0]
    for i in range(24):
        int_x = mdm_jts[:, i, 0]
        int_y = mdm_jts[:, i, 1]
        int_z = mdm_jts[:, i, 2]
        x = np.arange(0, N)
        f_x = interpolate.interp1d(x, int_x)
        f_y = interpolate.interp1d(x, int_y)
        f_z = interpolate.interp1d(x, int_z)
        
        new_x = f_x(np.linspace(0, N-1, int(N * 1.5)))
        new_y = f_y(np.linspace(0, N-1, int(N * 1.5)))
        new_z = f_z(np.linspace(0, N-1, int(N * 1.5)))
        jts.append(np.stack([new_x, new_y, new_z], axis = 1))
    jts = np.stack(jts, axis = 1)
    return jts

```

closed 2023-12-31 by ZhengyiLuo

huey0528, 2024-01-03:
> phc_shape_mcp_iccv-2023-12-06-14_22_24.mov 
>  phc_shape_mcp_iccv-2023-12-06-14_11_02.mov

Hi, I am fail to generate a good motion clip with mdm, may I know which kind of data you transfer from mdm to this github? I tried both the mdm generated npy file, and the render_mesh generated npy file. Thanks ~

huey0528, 2024-01-03:
> Hello! I've tried your model with some motion clips generated by mdm. However, the model is not doing well on theses clips. The following videos are some examples.
> 
>  phc_shape_mcp_iccv-2023-12-06-14_22_24.mp4 
> 
>  phc_shape_mcp_iccv-2023-12-06-14_11_02.mp4 
> 
> Is it because of the low quality of the motion generated by mdm or other reasons? Do you have any suggestions to improve? Thank you!

Hi, may I know how you transfer the data from mdm to phc?

ZhengyiLuo, 2024-01-08:
@huey0528  language-based demo using MDM has been released, see [docs](https://github.com/ZhengyiLuo/PerpetualHumanoidControl/blob/master/docs/language_to_control_demo.md) for instructions.

huey0528, 2024-01-08:
> @huey0528 language-based demo using MDM has been released, see [docs](https://github.com/ZhengyiLuo/PerpetualHumanoidControl/blob/master/docs/language_to_control_demo.md) for instructions.

Thanks.

Cr7st, 2024-01-08:
> Hi, may I know how you transfer the data from mdm to phc?

There is script in MDM to transfer its result to smpl parameters. Just further transfer the up-axis of the rotation data. And also many thanks to @ZhengyiLuo for releasing language-based demo

Jatinkalal, 2024-12-12:
@Cr7st , hey could you tell more about  ? (also which script)  as generally MDM outputs 3d keypoints right.
Thanks.

zhuyaoye, 2025-06-20:
> Hello! I've tried your model with some motion clips generated by mdm. However, the model is not doing well on theses clips. The following videos are some examples. [...]

Hi! Could you please tell me your prompt to generate the first video? Thanks.