ZhengyiLuo/PHC #18: cant find the basicmodel_neutral_lbs_10_207_0_v1.1.0, basicmodel_m_lbs_10_207_0_v1.1.0.pkl, basicmodel_f_lbs_10_207_0_v1.1.0.pkl  from the link i wish you would put the exact download link

mostafablender, 2023-12-31:
cant find the basicmodel_neutral_lbs_10_207_0_v1.1.0, basicmodel_m_lbs_10_207_0_v1.1.0.pkl, basicmodel_f_lbs_10_207_0_v1.1.0.pkl  from the link i wish you would put the exact download link 

also a short video about how to download and install it would help 

also i had to change 'requirement'  to "requirements"  for this to work

i still need help

ZhengyiLuo, 2023-12-31:
Hi @mostafablender, 

The exact download link would be [https://download.is.tue.mpg.de/download.php?domain=smpl&sfile=SMPL_python_v.1.1.0.zip](https://download.is.tue.mpg.de/download.php?domain=smpl&sfile=SMPL_python_v.1.1.0.zip), though this is behind a login wall for the SMPL model. I do not own the SMPL body models so it has to be downloaded from the official website. 

Thanks about the `requirement.txt` file, it has been fixed.

mostafablender, 2024-01-01:
thanks i may ask question again soon  i may find out problem  again ,  :)

closed 2024-01-01 by mostafablender