ZhengyiLuo/PHC #19: what i do lol

mostafablender, 2024-01-01:
i went this stpes to install it 
1- download and unzip   the `PerpetualHumanoidControl-master`  and in the folder i open cmd  and  then i wrote  this  to work 
conda create -n isaac python=3.8
conda install pytorch torchvision torchaudio pytorch-cuda=11.6 -c pytorch -c nvidia
pip install -r requirements.txt  

2- i went to nvidia from the link  to download it  . clicked on the join button  "omniverse" installed  ( i installed it in different folder as nvidia wanted to install it , i have no idea if this was the thing i had installed ) 

3- i downloaded the 3 file and renamed them as you said  . but now  i dont know where exactly i have to put them lol   ????
i have no idea

ZhengyiLuo, 2024-01-01:
For [Isaac gym](https://developer.nvidia.com/isaac-gym), click the "join now" button and login: 
[image]
which will take you to a download page. 

Isaac Gym is different from the Omniverse stack and doesn't have an installer. 

For the three renamed files, put them in the `data/smpl` folder.

closed 2024-01-08 by ZhengyiLuo

mostafablender, 2024-01-12:
Hi again  is Isaac Gym  only for Linux ?  i cant find  it for windows :/

kexul, 2024-01-12:
> Hi again is Isaac Gym only for Linux ? i cant find it for windows :/

I think so. Even WSL cannot run it : https://forums.developer.nvidia.com/t/isaac-gym-on-windows-subsystem-for-linux-wsl/201208

mostafablender, 2024-01-12:
> > Hi again is Isaac Gym only for Linux ? i cant find it for windows :/
> 
> I think so. Even WSL cannot run it : https://forums.developer.nvidia.com/t/isaac-gym-on-windows-subsystem-for-linux-wsl/201208

i dont even have enough space for wsl to install it . i just want to try it and see if i can get good animation result to import it into blender 3d