ZhengyiLuo/PHC #22: Question about required motion_file (amass_isaac_im_train_take6_upright_slim.pkl) missing in training training_prim step  in section "Training PHC"

Feiyu-Yao, 2024-01-11:
Hi, 

The first step in section "Training PHC", aka training training_prim step, use the following command:
'''
python phc/run.py --task HumanoidIm --cfg_env phc/data/cfg/phc_shape_pnn_train_iccv.yaml --cfg_train phc/data/cfg/train/rlg/im_pnn.yaml --motion_file data/amass/pkls/amass_isaac_im_train_take6_upright_slim.pkl --network_path output/phc_shape_pnn_train_iccv --headless --has_eval
'''
However, the file "data/amass/pkls/amass_isaac_im_train_take6_upright_slim.pkl" seems missing in folder, would mind share how this file generated?

ZhengyiLuo, 2024-01-11:
Hi! 

For training data, please follow the [processing amass data](https://github.com/ZhengyiLuo/PerpetualHumanoidControl#data-processing-amass) section for downloading and processing the AMASS dataset. I do not own the amass dataset so I can not redistribute the processed data🙏🙏🙏

Feiyu-Yao, 2024-01-11:
Thanks for your response. 

I have run the data process script. There is no file named "data/amass/pkls/amass_isaac_im_train_take6_upright_slim.pkl". But there exists "data/amass/pkls/amass_copycat_take5_train.pkl". The data processing script "convert_amass_isaac.py" sets "--out_file" be data/amass/pkls/amass_copycat_take5_train.pkl. 

I think the two files are actually means the same thing and should be consistent.

ZhengyiLuo, 2024-01-11:
Oh, my bad. The output of the `conver_amass_isaac.py` is what you need for training. I have updated the code for the output file name.

closed 2024-01-11 by ZhengyiLuo