ZhengyiLuo/PHC #23: Regarding HumanoidImGetup

EvelynFan, 2024-01-12:
Thanks for the great work! In the section of Training PHC, for HumanoidImGetup, the motion files for training primitive net 3 and the composer are different. What's the difference? I only saw "amass_isaac_standing_upright_slim.pkl" in the sample_data folder. This file only contains 1000 frames. Does the motion file "data/amass/pkls/amass_isaac_run_take6_upright_slim.pkl" contain more frames?

ZhengyiLuo, 2024-02-07:
The `amass_isaac_standing_upright` file is only there for demo purposes; it only contains a standing motion. 

For training, the full AMASS dataset will need to be processed. The `amass_isaac_run_take6_upright_slim` is a subset of AMASS that contains mostly locomotion.

EvelynFan, 2024-02-07:
> The `amass_isaac_standing_upright` file is only there for demo purposes; it only contains a standing motion. 
> 
> 
> 
> For training, the full AMASS dataset will need to be processed. The `amass_isaac_run_take6_upright_slim` is a subset of AMASS that contains mostly locomotion. 

Thanks Zhengyi.

closed 2024-02-07 by EvelynFan