ZhengyiLuo/PHC #26: A few minor questions

ZDDWLIG, 2024-02-18:
Thank you for your impressive work! I am very interested in your work but I have encountered some problems while trying to reproduce it. The problems are as follows:
1. Which command can get a video of a humanoid in continuous motion, such as boxing shown in your video.
2. What is the approximate GPU memory required for inference and training. I evaluated it on AMASS on a 6g 1660 super, but got an out-of-memory error from isaac gym.

[image]

Hope you can give some instructions. Thanks a lot.

ZhengyiLuo, 2024-02-18:
Thanks for you interest in my work!

For the boxing motion, please refer to the [webcam demo](https://github.com/ZhengyiLuo/PerpetualHumanoidControl/blob/master/docs/video_to_control_demo.md) and [language demo](https://github.com/ZhengyiLuo/PerpetualHumanoidControl/blob/master/docs/language_to_control_demo.md) 

For inference from my end it looks like it takes around 4G memory, though I think 6G might be a little hard. Try reducing the memory buffer [here](https://github.com/ZhengyiLuo/PerpetualHumanoidControl/blob/93ed068fb4228193c61ef0076e491f07356d119c/phc/utils/config.py#L190).

ZDDWLIG, 2024-02-27:
Thank you so much for the detailed clarifications.

closed 2024-03-01 by ZhengyiLuo