ZhengyiLuo/PHC #28: hydra error

GuyTevet, 2024-02-22:
Hi @ZhengyiLuo . Thank you for your great work!

I'm running according to the README:
```
python phc/run_hydra.py learning=im_big learning.params.network.ending_act=False exp_name=phc_comp_kp_2 env.obs_v=7 env=env_im_getup_mcp robot=smpl_humanoid robot.real_weight_porpotion_boxes=False env.motion_file=sample_data/amass_isaac_standing_upright_slim.pkl env.models=['output/HumanoidIm/phc_kp_2/Humanoid.pth'] env.num_prim=3 env.num_envs=1  headless=False epoch=-1 test=True
```

And getting:
```
Could not override 'learning.params.network.ending_act'.
To append to your config use +learning.params.network.ending_act=False
Key 'ending_act' is not in struct
    full_key: learning.params.network.ending_act
    object_type=dict

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```
 I'm probably missing something basic, can you take a look? 
Thanks!

GuyTevet, 2024-02-22:
The next fail was:
```
Error(s) in loading state_dict for Network:
	Unexpected key(s) in state_dict: "a2c_network.composer.0.weight", "a2c_network.composer.0.bias", "a2c_network.composer.2.weight", "a2c_network.composer.2.bias", "a2c_network.composer.4.weight", "a2c_network.composer.4.bias", "a2c_network.composer.6.weight", "a2c_network.composer.6.bias", "a2c_network.composer.8.weight", "a2c_network.composer.8.bias", "a2c_network.composer.10.weight", "a2c_network.composer.10.bias", "a2c_network.composer.12.weight", "a2c_network.composer.12.bias". 
  File "/home/dcor/guytevet/PHC/phc/learning/amp_players.py", line 65, in restore
    super().restore(fn)
  File "/home/dcor/guytevet/PHC/phc/run_hydra.py", line 336, in main
    runner.run(cfg)
  File "/home/dcor/guytevet/PHC/phc/run_hydra.py", line 342, in <module>
    main()
RuntimeError: Error(s) in loading state_dict for Network:
	Unexpected key(s) in state_dict: "a2c_network.composer.0.weight", "a2c_network.composer.0.bias", "a2c_network.composer.2.weight", "a2c_network.composer.2.bias", "a2c_network.composer.4.weight", "a2c_network.composer.4.bias", "a2c_network.composer.6.weight", "a2c_network.composer.6.bias", "a2c_network.composer.8.weight", "a2c_network.composer.8.bias", "a2c_network.composer.10.weight", "a2c_network.composer.10.bias", "a2c_network.composer.12.weight", "a2c_network.composer.12.bias".
```

Probably since `network_path='output/HumanoidIm/phc_comp_kp_2'` instead of `output/HumanoidIm/phc_kp_2` but I didn't understand how to configure it through hydra.

merc-s, 2024-02-23:
There seems to be a mismatch between the loaded model architecture, the chekpoints, and the learning.yaml config file. The key errors occur because the model network either has one or multiple primitives, while the opposite is true for the checkpoint.

I managed to get through the key errors by setting `learning=im_pnn_big`, `env.models=['output/HumanoidIm/phc_kp_2/Humanoid.pth']`.

After changing this, a new shape mismatch error occurs, where the model's output is 3 while the checkpoint's output is 69. Manually changing the model's output to 69 during runtime finally lets me load the model. Unfortunately, I get the following error:
`
Traceback (most recent call last):
  File "phc/run_hydra.py", line 339, in main
    runner.run(cfg)
  File "/home/merkourios/miniconda3/envs/isaac/lib/python3.8/site-packages/rl_games/torch_runner.py", line 144, in run
    player.run()
  File "/home/merkourios/code/PHC/phc/learning/im_amp_players.py", line 259, in run
    action = self.get_action(obs_dict, is_determenistic)
  File "/home/merkourios/code/PHC/phc/learning/common_player.py", line 157, in get_action
    output = super().get_action(obs_dict['obs'], is_determenistic)
  File "/home/merkourios/miniconda3/envs/isaac/lib/python3.8/site-packages/rl_games/algos_torch/players.py", line 65, in get_action
    return rescale_actions(self.actions_low, self.actions_high, torch.clamp(current_action, -1.0, 1.0))
  File "/home/merkourios/miniconda3/envs/isaac/lib/python3.8/site-packages/rl_games/algos_torch/players.py", line 14, in rescale_actions
    scaled_action =  action * d + m
RuntimeError: The size of tensor a (69) must match the size of tensor b (3) at non-singleton dimension 1
`

So it seems the action space should've been 3 after all. Maybe the checkpoint has a problem?

GuyTevet, 2024-02-24:
I'm just running your line from the readme without any changes.

ZhengyiLuo, 2024-02-24:
My bad! 

The correct command should be : 
```
python phc/run_hydra.py learning=im_mcp_big learning.params.network.ending_act=False exp_name=phc_comp_kp_2 env.obs_v=7 env=env_im_getup_mcp robot=smpl_humanoid robot.real_weight_porpotion_boxes=False env.motion_file=sample_data/amass_isaac_standing_upright_slim.pkl env.models=['output/HumanoidIm/phc_kp_2/Humanoid.pth'] env.num_prim=3 env.num_envs=1  headless=False epoch=-1 test=True
```

ZhengyiLuo, 2024-02-24:
Since the MCP  net really is just an MLP with an output of 3, the newer models I train I use the im_big learning config instead of the MCP nets (with the composer key), thus the confusion on my part in releasing this model. Should be resolved now!

closed 2024-02-24 by ZhengyiLuo

GuyTevet, 2024-02-24:
Working. Thanks @ZhengyiLuo !