ZhengyiLuo/PHC #30: Blank display

GuyTevet, 2024-02-25:
Hi @ZhengyiLuo ,
Running PHC  evaluations, e.g.:
```
python phc/run_hydra.py learning=im_mcp_big learning.params.network.ending_act=False exp_name=phc_comp_kp_2 env.obs_v=7 env=env_im_getup_mcp robot=smpl_humanoid robot.real_weight_porpotion_boxes=False env.motion_file=sample_data/amass_isaac_standing_upright_slim.pkl env.models=['output/HumanoidIm/phc_kp_2/Humanoid.pth'] env.num_prim=3 env.num_envs=1  headless=False epoch=-1 test=True
```

The display (Xephyr) remains blank.
As a sanity check,
```
python scripts/joint_monkey_smpl.py
```
works properly.

ZhengyiLuo, 2024-02-26:
Try adding the `no_virtual_display=True` flag. 
```
python phc/run_hydra.py learning=im_mcp_big learning.params.network.ending_act=False exp_name=phc_comp_kp_2 env.obs_v=7 env=env_im_getup_mcp robot=smpl_humanoid robot.real_weight_porpotion_boxes=False env.motion_file=sample_data/amass_isaac_standing_upright_slim.pkl env.models=['output/HumanoidIm/phc_kp_2/Humanoid.pth'] env.num_prim=3 env.num_envs=1  headless=False epoch=-1 test=True no_virtual_display=True
```

The virtual display stuff is super good for recording the viewer (isaac gym does not provide such function, it only allows for recording a single environment), but it can be finicky.

GuyTevet, 2024-02-26:
Yup that resolves it. Thanks!

closed 2024-02-26 by GuyTevet