ZhengyiLuo/PHC #33: Training: config files

merc-s, 2024-02-29:
Hi @ZhengyiLuo,

I am trying to replicate the PNN training process and I was wondering which config files are the most recent. I understand this is a very active project; are the README instructions up-to-date?

Thanks in advance.

ZhengyiLuo, 2024-03-01:
To train the PNN model, the newest command would be 
```
phc/run_hydra.py learning=im_pnn_big env=env_im robot=smpl_humanoid env.motion_file=[motion_file] exp_name=[exp_name]
```

And then train the MCP model 

```
python phc/run_hydra.py learning=im_mcp_big  exp_name=[exp_name] env=env_im_getup_mcp robot=smpl_humanoid env.zero_out_far=False robot.real_weight_porpotion_boxes=False env.num_prim=3 env.motion_file=[motion_file] env.models=['models']
```

And follow the procedure for data processing and advancing the PMCP process.

ZhengyiLuo, 2024-03-01:
This will replicate the training process the one used in [PULSE](https://www.zhengyiluo.com/UniversalMotionRepresentation/), which is the best one I have for now.

merc-s, 2024-03-01:
Running the command with the above config files gives me the following error:
```
Traceback (most recent call last):
  File "phc/run_hydra.py", line 339, in main
    runner.run(cfg)
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/torch_runner.py", line 139, in run
    self.run_train()
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/torch_runner.py", line 125, in run_train
    agent.train()
  File "/home/gymuser/src/phc/learning/common_agent.py", line 121, in train
    train_info = self.train_epoch()
  File "/home/gymuser/src/phc/learning/amp_agent.py", line 424, in train_epoch
    batch_dict = self.play_steps()
  File "/home/gymuser/src/phc/learning/amp_agent.py", line 328, in play_steps
    res_dict = self.get_action_values(self.obs)
  File "/home/gymuser/src/phc/learning/common_agent.py", line 275, in get_action_values
    res_dict = self.model(input_dict)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1480, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/gymuser/src/phc/learning/amp_models.py", line 28, in forward
    result = super().forward(input_dict)
  File "/home/gymuser/.local/lib/python3.8/site-packages/rl_games/algos_torch/models.py", line 227, in forward
    mu, logstd, value, states = self.a2c_network(input_dict)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1480, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/gymuser/src/phc/learning/amp_network_builder.py", line 45, in forward
    actor_outputs = self.eval_actor(obs_dict)
  File "/home/gymuser/src/phc/learning/amp_network_pnn_builder.py", line 67, in eval_actor
    a_out, a_outs = self.pnn(a_out, idx=self.training_prim)
  File "/usr/local/lib/python3.8/dist-packages/torch/nn/modules/module.py", line 1480, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/gymuser/src/phc/learning/pnn.py", line 98, in forward
    assert len(curr_actor) == 5  # Only support three MLPs right now
AssertionError
```
which I fixed by adding ```has_lateral:False``` inside ```env_im```.

At this point I am able to start training and I do see an increase in ```mean_reward```, so all good so far. However, I am still not quite sure on how I should proceed. The README instructions are not very clear to me because they reference different config files.

For instance, how do I move to the next primitive? Is it the following flags? Where should the current model be stored to continue training?
```
epoch=[include epoch+1 from previous step] env.fitting=True env.training_prim=1
```

ZhengyiLuo, 2024-03-01:
Ahh I see, sorry my bad, missed one command here:

```
phc/run_hydra.py learning=im_pnn_big env=env_im_pnn robot=smpl_humanoid env.motion_file=[motion_file] exp_name=[exp_name]

```

Should use the `env=env_im_pnn` basically

ZhengyiLuo, 2024-03-01:
I have updated the instructions to the newest released model configs; could you see if it works? 

You should expect the success rate to reach better than  90% pretty soon (say 6000 epoches?).

merc-s, 2024-03-01:
Thank you, I will try as soon as possible and I will let you know. Marking as closed for now!

closed 2024-03-01 by merc-s