ZhengyiLuo/PHC #34: UHC vs PHC steps per second discrepancy

kinalmehta, 2024-03-01:
**UHC**
number of substeps = 15 [link](https://github.com/ZhengyiLuo/UHC/blob/2419deef54958b47b0a0cbf0aee50c82426e7950/uhc/envs/humanoid_im.py#L112)
mujoco simulation time = 0.002222 [link](https://github.com/ZhengyiLuo/UHC/blob/2419deef54958b47b0a0cbf0aee50c82426e7950/assets/mujoco_models/humanoid_smpl_neutral_mesh.xml#L13)
steps per second = 1/(simulation time * number of substeps) = 30

**PHC**
number of substeps = 2 [link](https://github.com/ZhengyiLuo/PHC/blob/148fa629195f03fe1fa8c0635efe3562faac163e/phc/data/cfg/env/phc_kp_mcp_iccv.yaml#L63) _notice that the comment in the config file mentions 30 Hz_
physx simulation time = 0.0055 [link](https://github.com/ZhengyiLuo/PHC/blob/148fa629195f03fe1fa8c0635efe3562faac163e/test_good.xml#L4)
steps per second = 1/(simulation time * number of substeps) = 90

Moreover I also notice that in UHC the target frame is the next frame based on the frames passed, whereas in PHC the target frame idx is computed using a logic which I couldn't comprehend.

Could you please help me understand the reason for this difference?

ZhengyiLuo, 2024-03-01:
PHC is using simulation time is [1/60](https://github.com/ZhengyiLuo/PHC/blob/148fa629195f03fe1fa8c0635efe3562faac163e/phc/utils/config.py#L47), which is hardcoded in. With substep is 2, it's 30 HZ. 

PHC it's the same, [one frame in the future](https://github.com/ZhengyiLuo/PHC/blob/148fa629195f03fe1fa8c0635efe3562faac163e/phc/env/tasks/humanoid_im.py#L696)

For rendering, I rendered PHC in 60 HZ, and dump the video in 60HZ.

kinalmehta, 2024-03-05:
Thanks for the clarification. Missed this somehow.

closed 2024-03-05 by kinalmehta