ZhengyiLuo/PHC #38: how to get quantitative results ?

siyuzou, 2024-03-07:
Hi @ZhengyiLuo, thanks for the great work !

Can you provide with some scripts that test models quantitatively and output those metrics (e.g., g-mpjpe, acc) ? 

I can only find `phc/run_hydra.py`, and seems it only has training and inference (playing) abilities. Did I missed something ?

ZhengyiLuo, 2024-03-09:
Please refer to the [evaluation](https://github.com/ZhengyiLuo/PHC?tab=readme-ov-file#evaluation) section. 

Specifically, "evaluate on full AMASS". The flag "im_eval" will iterate through the dataset and compute metrics.

siyuzou, 2024-03-11:
Thanks for the reply !

Now I can do quantitative evaluations wIth `im_eval` and `headless` flags on. `headless` should stay true or error occurs.

Thanks again :)

closed 2024-03-11 by siyuzou