ZhengyiLuo/PHC #44: ModuleNotFoundError: No module named 'smpl_sim.utils'

kunkun0w0, 2024-04-06:
Thanks for releasing the PHC!!!!!

I met a problem when I tried to run `python scripts/vis/vis_motion_mj.py`
[image]

I also checked the `smpl_sim` and found no `utils` file or folder.
[image]

I used the dockerfile in PHC.

kunkun0w0, 2024-04-07:
I fixed it by manually copying the lost folder.

closed 2024-04-07 by kunkun0w0

xindonglin99, 2024-04-11:
Another thing that you can do is to install the smpl_sim with
```pip install -e git+https://github.com/ZhengyiLuo/SMPLSim.git@master#egg=smpl_sim```
See [this](https://stackoverflow.com/questions/10847764/after-installation-of-python-package-from-git-repository-some-folders-are-missin) for reference.