ZhengyiLuo/PHC #46: Filtering sequences for AMASS w/ SMPL-X

areiner222, 2024-04-15:
Hi,

Thanks for this great work!

First a quick question on amass occlusions, I noticed that some of the sequences map to an issue of "normal" - are those meant to be excluded? For instance, I see sequences on a treadmill marked as "normal" but assume they should not be included?

Also, for AMASS have you gone through the smpl-x annotations to find corrupted, unsuitable, corrupted sequences like you did for smpl? 
If not, I was thinking of going through that process and it would be great to get any advice / pointers on what your process was like to identify the sequences you removed.

Thank you!

ZhengyiLuo, 2024-04-29:
"normal" means sequences recorded on a treadmill; their position in some way is "normalized". Sorry for the bad name. Yes they should be excluded. 

From smpl-x I have done a similar process; but it's still ongoing (as there are many more issues with the amass dataset using smplx). I will release it once I have finished the process. Stay tuned!

closed 2024-05-14 by ZhengyiLuo

areiner222, 2024-08-18:
Did you ever manage to curate a list of viable sequences for AMASS SMPL-X? I'm hopefully you might have given PHC-X in the works! 🤞 

Thanks again!