ZhengyiLuo/PHC #50: How should I solve it?

zhangqingbo6, 2024-05-14:
Hello! I encountered "Traceback (most recent call last)" when running "Running the video to control demo":
   File "phc/run.py", line 297, in <module>
     main()
   File "phc/run.py", line 291, in main
     runner.run(vargs)
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/torch_runner.py", line 142, in run
     player = self.create_player()
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/torch_runner.py", line 128, in create_player
     return self.player_factory.create(self.algo_name, config=self.config)
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/common/object_factory.py", line 15, in create
     return builder(**kwargs)
   File "phc/run.py", line 209, in <lambda>
     runner.player_factory.register_builder('im_amp', lambda **kwargs: im_amp_players.IMAMPPlayerContinuous(**kwargs))
   File "/home/ubuntu/issac-test/PHC-master/phc/learning/im_amp_players.py", line 28, in __init__
     super().__init__(config)
   File "/home/ubuntu/issac-test/PHC-master/phc/learning/amp_players.py", line 23, in __init__
     super().__init__(config)
   File "/home/ubuntu/issac-test/PHC-master/phc/learning/common_player.py", line 16, in __init__
     BasePlayer.__init__(self, config)
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/common/player.py", line 17, in __init__
     self.env = self.create_env()
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/common/player.py", line 136, in create_env
     return env_configurations.configurations[self.env_name]['env_creator'](**self.env_config)
   File "phc/run.py", line 193, in <lambda>
     env_configurations.register('rlgpu', {'env_creator': lambda **kwargs: create_rlgpu_env(**kwargs), 'vecenv_type': 'RLGPU'})
   File "phc/run.py", line 88, in create_rlgpu_env
     task, env = parse_task(args, cfg, cfg_train, sim_params)
   File "/home/ubuntu/issac-test/PHC-master/phc/utils/parse_task.py", line 60, in parse_task
     task = eval(args.task)(cfg=cfg, sim_params=sim_params, physics_engine=args.physics_engine, device_type=args.device, device_id=device_id, headless=args.headless)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid_im_mcp_demo.py", line 38, in __init__
     super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid_im_mcp.py", line 22, in __init__
     super().__init__(cfg=cfg, sim_params=sim_params, physics_engine=physics_engine, device_type=device_type, device_id=device_id, headless=headless)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid_im.py", line 48, in __init__
     self.load_humanoid_configs(cfg)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid.py", line 247, in load_humanoid_configs
     self.humanoid_type = cfg.robot.humanoid_type
AttributeError: 'dict' object has no attribute 'robot'" How should I solve it?

ZhengyiLuo, 2024-05-14:
Hi! 

I was able to run the command on my end; did you pull the newest code?

zhangqingbo6, 2024-05-15:
Thank you very much for your quick answer! But following your instructions, a new problem seems to have arisen. My video_to_pose_server.py is opening and running normally.
”Traceback (most recent call last):
   File "phc/run_hydra.py", line 343, in <module>
     main()
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/main.py", line 94, in decorated_main
     _run_hydra(
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
     _run_app(
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/_internal/utils.py", line 457, in _run_app
     run_and_report(
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
     raise ex
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
     return fun()
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
     lambda: hydra.run(
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 132, in run
     _ = ret.return_value
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/core/utils.py", line 260, in return_value
     raise self._return_value
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/hydra/core/utils.py", line 186, in run_job
     ret.return_value = task_function(task_cfg)
   File "phc/run_hydra.py", line 337, in main
     runner.run(cfg)
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/torch_runner.py", line 144, in run
     player.run()
   File "/home/ubuntu/issac-test/PHC-master/phc/learning/im_amp_players.py", line 231, in run
     obs_dict = self.env_reset()
   File "/home/ubuntu/issac-test/PHC-master/phc/learning/common_player.py", line 193, in env_reset
     obs = self.env.reset(env_ids)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/vec_task_wrappers.py", line 55, in reset
     self.task.reset(env_ids)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid.py", line 511, in reset
     self._reset_envs(env_ids)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid_amp_task.py", line 74, in _reset_envs
     super()._reset_envs(env_ids)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid_amp.py", line 345, in _reset_envs
     super()._reset_envs(env_ids)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid.py", line 564, in _reset_envs
     self._compute_observations(env_ids)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid_im_mcp_demo.py", line 137, in _compute_observations
     task_obs = self._compute_task_obs_demo(env_ids)
   File "/home/ubuntu/issac-test/PHC-master/phc/env/tasks/humanoid_im_mcp_demo.py", line 254, in _compute_task_obs_demo
     json_data = pose_res.json()
   File "/home/ubuntu/anaconda3/envs/phc/lib/python3.8/site-packages/requests/models.py", line 975, in json
     raise RequestsJSONDecodeError(e.msg, e.doc, e.pos)
requests.exceptions.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
"

closed 2024-05-16 by zhangqingbo6