ZhengyiLuo/PHC #54: No module named 'smpl_sim.khrylib'

ElviraLL, 2024-06-19:
Thanks for this awesome work!

I was trying to process the SMPL data according to the provided document but met this error when importing `smpl_sim.khrylib`. I installed the environment using the requirement.txt file. Could you please take a look? Thanks!
```
  File "scripts/data_process/convert_amass_isaac.py", line 17, in <module>
    from smpl_sim.khrylib.utils import get_body_qposaddr
ModuleNotFoundError: No module named 'smpl_sim.khrylib'
```

Soumyabrata2003, 2024-06-19:
Comment that line out. It isn't used in the code anyway. Worked for me.

ElviraLL, 2024-06-19:
Thanks!

> Comment that line out. It isn't used in the code anyway. Worked for me.

closed 2024-06-19 by ElviraLL