ZhengyiLuo/PHC #55: Use my own robot structure, do I still need to retarget?

learning-github-zgy, 2024-06-21:
Thank you for your great work!
If I use my own robot structure, do I still need to retarget the smpl humanoid structure to the custom robot structure? How do I get this part of the job done? Can you give me a hint?

learning-github-zgy, 2024-06-21:
> Thank you for your great work!感谢你杰出的工作！ If I use my own robot structure, do I still need to retarget the smpl humanoid structure to the custom robot structure? How do I get this part of the job done? Can you give me a hint? 如果我使用自己的机器人结构，是否仍需要将 smpl 人形结构重新定位为自定义机器人结构？我如何完成这部分工作？你能给我一个提示吗？

> Thank you for your great work! If I use my own robot structure, do I still need to retarget the smpl humanoid structure to the custom robot structure? How do I get this part of the job done? Can you give me a hint?

Actually, I want to use the unitree_h1 robot

learning-github-zgy, 2024-06-21:
> Thank you for your great work! If I use my own robot structure, do I still need to retarget the smpl humanoid structure to the custom robot structure? How do I get this part of the job done? Can you give me a hint?

Actually, I want to use the unitree_h1 robot

ZhengyiLuo, 2024-06-27:
Yes! Retargeting would be needed. 

Checkout our work [H2O](https://human2humanoid.com/) and [OmniH2O](https://omni.human2humanoid.com/).

closed 2024-06-27 by ZhengyiLuo