ZhengyiLuo/PHC #57: Updated mocap and language demo

ElviraLL, 2024-07-15:
Hey thanks for the awsome work, I noticed that the settings and configuration for webcam and language demo were based on previous version of you implementation before introducing hydra. Would you update the codebase and instructions soon? Thanks

ElviraLL, 2024-07-15:
```
/home/PC/Documents/projs/PHC/phc/env/tasks/humanoid_im_mcp_demo.py:272: RuntimeWarning: invalid value encountered in true_divide
  ref_rb_pos /= scale[:, None, None]
Traceback (most recent call last):
  File "phc/run.py", line 297, in <module>
    main()
  File "phc/run.py", line 291, in main
    runner.run(vargs)
  File "/home/PC/miniconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/torch_runner.py", line 144, in run
    player.run()
  File "/home/PC/Documents/projs/PHC/phc/learning/im_amp_players.py", line 259, in run
    action = self.get_action(obs_dict, is_determenistic)
  File "/home/PC/Documents/projs/PHC/phc/learning/common_player.py", line 155, in get_action
    output = super().get_action(obs_dict['obs'], is_determenistic)
  File "/home/PC/miniconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/algos_torch/players.py", line 53, in get_action
    res_dict = self.model(input_dict)
  File "/home/PC/miniconda3/envs/rlgpu/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1532, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/PC/miniconda3/envs/rlgpu/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1541, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/PC/Documents/projs/PHC/phc/learning/amp_models.py", line 28, in forward
    result = super().forward(input_dict)
  File "/home/PC/miniconda3/envs/rlgpu/lib/python3.8/site-packages/rl_games/algos_torch/models.py", line 229, in forward
    distr = torch.distributions.Normal(mu, sigma)
  File "/home/PC/miniconda3/envs/rlgpu/lib/python3.8/site-packages/torch/distributions/normal.py", line 56, in __init__
    super().__init__(batch_shape, validate_args=validate_args)
  File "/home/PC/miniconda3/envs/rlgpu/lib/python3.8/site-packages/torch/distributions/distribution.py", line 68, in __init__
    raise ValueError(
ValueError: Expected parameter loc (Tensor of shape (1, 4)) of distribution Normal(loc: torch.Size([1, 4]), scale: torch.Size([1, 4])) to satisfy the constraint Real(), but found invalid values:
tensor([[nan, nan, nan, nan]], device='cuda:0')
```
I also met this error when running forward function

ZhengyiLuo, 2024-07-31:
Updated. Thanks!

closed 2024-07-31 by ZhengyiLuo