ZhengyiLuo/PHC #63: How to train a PHC on my own data

yw0208, 2024-08-18:
Hi,
I am trying to train a PHC with my data. But I don't want to train a brand new PHC but based on a pre-trained PHC. The following are the training steps designed based on my understanding:

1. I should train the next primitive based on the Humanoid.pth in phc_kp_pnn_iccv by running the following command.
`python phc/run_hydra.py learning=im_pnn_big env=env_im_pnn robot=smpl_humanoid env.motion_file=[motion_file] epoch=[include epoch+1 from previous step] env.fitting=True env.training_prim=1`

2. I should run the following command to dump hard sequences and a copied primitive. 
`python scripts/pmcp/forward_pmcp.py --exp [exp_name] --epoch [epoch] --idx {idx}`

3. Repeat steps 1 and 2.

4. Train a brand new composer by running the following command
`python phc/run_hydra.py learning=im_pnn_big exp_name=phc_shape_pnn_iccv env=env_im_pnn robot=smpl_humanoid env.motion_file=[motion_file] epoch=[include current epoch + 1 from the forward_pmcp step] env.fitting=True env.training_prim=[+1] env.zero_out_far=True env.zero_out_far_train=True env.getup_udpate_epoch={epoch}`

Are these steps right? I would appreciate it if you could point out any of my mistakes.

Thank you

ZhengyiLuo, 2024-08-21:
I take you would like to fine-tune a pretrained PHC on new data? There are a few ways:

1. You can just fine tune one primitive, and then use that primitive for motion imitation purposes. This primitive would not have the failure state recovery, but it would have the ability to imitate the new motions. 
2. The approach you listed above. The im_pnn_big PHC has three primitives, the first two are used to learn motion imitation and the last one for failure state recovery. I would recommend fine-tuning the first two and leave the last one (failure state recovery) fixed.

closed 2024-08-30 by ZhengyiLuo

Sarah816, 2024-12-24:
Hi @ZhengyiLuo ,
I'd like to finetune one primitive. Is there a pretrained model that we can download? 
I noticed that there's a phc_prim_vr model, but the num_obs for this model is 430. For a rotation+keypoint model I think num_obs should be 934?