ZhengyiLuo/PHC #66: Simulation Environment

liaodisen, 2024-09-10:
Hello Everyone,

I am new to robotics, and I am using MacOS and a remote machine with A6000 GPU available. I can run the script
`python scripts/joint_monkey_smpl.py`, and I used XQuartz to load the GUI window on my local machine, though it is lagging. However, when I run  `python scripts/vis/vis_motion_mj.py`, I had error:

> libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/home/d7liao/anaconda3/envs/isaac/lib/python3.8/site-packages/glfw/__init__.py:914: GLFWError: (65543) b'GLX: Failed to create context: BadValue (integer parameter out of range for operation)'
  warnings.warn(message, GLFWError)
ERROR: could not create window

Does anyone have better suggestion on how to run mujoco using remote machine and how to render isaac gym more fluently?
Thank you!

Anonnyyy, 2024-10-08:
请问你用的gym preview4吗

ZhengyiLuo, 2024-10-08:
I have had better luck with VNC than XQuartz; there is always some issue when I tried to use xservers from command line when GPU rendering is involved.

ZhengyiLuo, 2024-10-08:
> 请问你用的gym preview4吗

Yes, preview 4.

Anonnyyy, 2024-10-09:
Thank you  very much! : )
However i met some problems when running:
`python scripts/vis/vis_motion_mj.py`
it shows:

>****************************** Current motion keys ******************************
Sampling motion: tensor([0])
standing
>*********************************************************************************
>0it [00:00, ?it/s]pbar 0it [00:00, ?it/s]
0it [00:00, ?it/s]
Traceback (most recent call last):
  File "scripts/vis/vis_motion_mj.py", line 105, in <module>
    motion_lib.load_motions(skeleton_trees=[sk_tree] * num_motions, gender_betas=[torch.zeros(17)] * num_motions, limb_weights=[np.zeros(10)] * num_motions, random_sample=False, start_idx=curr_start)
  File "/home/anony/omnigrasep/PHC/phc/utils/motion_lib_base.py", line 249, in load_motions
    res_acc.update(self.load_motion_with_skeleton(*jobs[0], None, 0))
  File "/home/anony/omnigrasep/PHC/phc/utils/motion_lib_smpl.py", line 116, in load_motion_with_skeleton
    for f in pbar:
  File "/home/anony/anaconda3/envs/rlgpu/lib/python3.8/site-packages/tqdm/std.py", line 1181, in __iter__
    for obj in iterable:
TypeError: 'int' object is not iterable

看起来像是motion import失败了， `motion_file = "sample_data/amass_isaac_standing_upright_slim.pkl"` 导入的motion list是空的。请问您有遇到这样的问题吗？

ZhengyiLuo, 2024-10-20:
> Thank you very much! : ) However i met some problems when running: `python scripts/vis/vis_motion_mj.py` it shows:
> 
> > ****************************** Current motion keys ******************************
> > Sampling motion: tensor([0])
> > standing
> > 
> > 0it [00:00, ?it/s]pbar 0it [00:00, ?it/s]
> > 0it [00:00, ?it/s]
> > Traceback (most recent call last):
> > File "scripts/vis/vis_motion_mj.py", line 105, in 
> > motion_lib.load_motions(skeleton_trees=[sk_tree] * num_motions, gender_betas=[torch.zeros(17)] * num_motions, limb_weights=[np.zeros(10)] * num_motions, random_sample=False, start_idx=curr_start)
> > File "/home/anony/omnigrasep/PHC/phc/utils/motion_lib_base.py", line 249, in load_motions
> > res_acc.update(self.load_motion_with_skeleton(*jobs[0], None, 0))
> > File "/home/anony/omnigrasep/PHC/phc/utils/motion_lib_smpl.py", line 116, in load_motion_with_skeleton
> > for f in pbar:
> > File "/home/anony/anaconda3/envs/rlgpu/lib/python3.8/site-packages/tqdm/std.py", line 1181, in **iter**
> > for obj in iterable:
> > TypeError: 'int' object is not iterable
> 
> 看起来像是motion import失败了， `motion_file = "sample_data/amass_isaac_standing_upright_slim.pkl"` 导入的motion list是空的。请问您有遇到这样的问题吗？

I just tried `sample_data/amass_isaac_standing_upright_slim.pkl` and it does contain the motion. Please pull the most recent code.

closed 2024-12-19 by ZhengyiLuo