ZhengyiLuo/PHC #71: Performance Issue with video_to_pose_server.py

ruoyiqiao, 2024-09-28:
Hi, thanks for sharing this fascinating work!

I have been testing the video_to_pose_server.py script on an NVIDIA 3090 GPU, but I am only achieving around 15 FPS, which is lower than the 30 FPS mentioned in your documentation. I am curious to know if TensorRT is a mandatory requirement to achieve the desired 30 FPS performance, or if there might be some other  reasons leading to this discrepancy.

I appreciate any insights you could provide on this!

ZhengyiLuo, 2024-10-08:
I didn't do tensorRT but I did use two machines to do the demo, one to run pose estimation and another one for simulation. Are you running simulation + pose estimation on the same GPU?

ruoyiqiao, 2024-10-10:
Yes, I'm running both simulation and pose estimation on the same GPU. After switching to a device with an A6000, it managed to reach 40 FPS (which is a bit unexpected) and I think maybe bottleneck is on the CPU side. Thank you for the clarification!

closed 2024-10-10 by ruoyiqiao