ZhengyiLuo/PHC #74: Questions about Humanoid Robot Policy

zengweishuai, 2024-10-13:
Hi Luo, thank you for releasing code for Unitree Robot!!!

For the Unitree policies, are they keypoint-based or rotation-based? If they are rotation-based, is there any way I can convert them into keypoint-based?

Expect your reply!

ZhengyiLuo, 2024-10-14:
Hi, 

1. I did not release trained models with them but only the code support. You can change keypoint-based and rotation-based based on your need (obs_v=7 is keypoint based and obs_v=6 is rotation based). 
2. No domain randomization. I have started adding DR in the humanoid_teleop.py file but haven't tested in real-world yet. 
3. For deploying on the real-humanoid, please refer to policy released in [OmniH2O](https://github.com/LeCAR-Lab/human2humanoid).

zengweishuai, 2024-11-20:
Thanks for your answer! This has helped a lot~

closed 2024-11-20 by zengweishuai