ZhengyiLuo/PHC #77: Questions Regarding H1 and G1 Support

TongZhangTHU, 2024-10-15:
Hi,

Thank you for releasing support for the Unitree H1 and G1 humanoid robots! I have a few questions regarding the implementation:

(1) In the Imitation H1 and G1 subsection, the script you provided is placed under the Evaluation section, but from reviewing the code, it appears to be for training rather than evaluation. Could you confirm if this is correct or if I've misunderstood something?
(2) While running the script in Imitation H1 and G1 subsection, I couldn’t find any part of the code explicitly making use of retargeting. Are you using retargeted data in these examples? (The only related reference I found was scripts/data_process/fit_smpl_motion.py, which seems to handle retargeting, but it doesn’t appear to be used in the script above. )
(3) Does the current code support real-time control of H1 and G1 using a webcam? If not, are there any plans to add this functionality?

Thanks again for the support and looking forward to your response!

ZhengyiLuo, 2024-10-20:
1) Sorry about that, I have moved it to the training section. 

2) The retargeting code is included but I haven't wrote documentation for it. Here is something to try 

```
python scripts/data_process/fit_smpl_motion.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +fit_all=True
python scripts/data_process/fit_smpl_shape.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +fit_all=True
python scripts/vis/vis_q_mj.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +motion_name=[name_1, name_2, etc]
```

3). In theory yes. I haven't fully connected the pipelines though.

TongZhangTHU, 2024-10-24:
Thank you for your response!

ZhengyiLuo, 2024-12-10:
Added documentation for retargeting in the most recent commit.