ZhengyiLuo/PHC #79: Question about method _refresh_sim_tensors()

Zijing5, 2024-10-22:
Hello! 
When I execute **post_physics_step** method in the class Humanoid(BaseTask), there is a method called **self._refresh_sim_tensors**, 
[image]
and then I step into the method, it jumps to the definition in the class HumaniodAMP(Humanoid) instead of that in the class Humanoid(BaseTask) itself.
[image]
[image]
I do not understand why it jumps to the definition in its son class.

ZhengyiLuo, 2024-11-08:
Hi! Since the child's class overrides the parent's class, the child's function is called.

Zijing5, 2024-11-19:
Thanks very much!