ZhengyiLuo/PHC #80: Question about H1/G1 robot evaluation

Perkins729, 2024-10-23:
Is there currently standard evaluation scripts for g1/h1?

Perkins729, 2024-10-23:
I have modified the former scripts for evaluation but failed 😢

ZhengyiLuo, 2024-11-08:
The same script plus im_eval=True test=True env.num_envs=[] would work for H1/G1.

dbdxnuliba, 2024-11-10:
@ZhengyiLuo ,could you please give the full command to evaluate g1/h1,thanks

ZhengyiLuo, 2024-12-10:
```
python phc/run_hydra.py  project_name=Robot_IM   robot=unitree_h1      env=env_im_h1_phc env.motion_file=sample_data/sample_dance_h1.pkl learning=im_pnn_big   exp_name=unitree_h1_pnn_realsim_092924 sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7 epoch=-1 test=True env.num_envs=1  headless=False 

```

closed 2024-12-10 by ZhengyiLuo