ZhengyiLuo/PHC #84: question on adding a new robot

luoye2333, 2024-11-04:
Hi author Luo! Thanks for your great works.
I am working on training phc on a new robot structure. I have added urdf, xml, meshes in phc/data/assets and i am working on adding config files in phc/cfg/robot. Is there a full workflow to add a new robot? Much appreciated if you could write some in the readme file.

Btw, is A100 necessary for training? How much gpu memory does it takes? Is it possible to train on a single 4090?

luoye2333, 2024-11-05:
Finished creating a new .yaml file, but i dont know the meaning of "extend config". Does it mean config the fixed links (e.g. the head and hand links) in "extend config"?

I am using the dance_sample_g1.pkl and getting error:
```
phc/env/tasks/humanoid_amp.py", line 625, in _set_env_state
    self._rigid_body_pos[env_ids] = rigid_body_pos
RuntimeError: shape mismatch: value tensor of shape [3072, 26, 3] cannot be broadcast to indexing result of shape [3072, 29, 3]
```
seems i need to create a new reference data to make body numbers and indexes correct, or the problem is in the .yaml file?

luoye2333, 2024-11-05:
Found the method to retarget the dataset (Sorry didnt see other issues first): https://github.com/ZhengyiLuo/PHC/issues/77#issuecomment-2425226165

IlanElfen, 2024-11-05:
Hi I'm also confused on how you retarget a different robot. I'm using [human2humanoid](https://github.com/LeCAR-Lab/human2humanoid), which uses PCH to retarget to a different robot. I'll admit I'm a little lost. Could you please walk me through the files you added, changed, etc. Thank you very much!

luoye2333, 2024-11-05:
> Hi I'm also confused on how you retarget a different robot. I'm using [human2humanoid](https://github.com/LeCAR-Lab/human2humanoid), which uses PCH to retarget to a different robot. I'll admit I'm a [...]

1. add .urdf, .xml and /meshes to phc/data/assets/robot
2. add a yaml file like unitree_g1.yaml in phc/data/cfg/robot
3. add a env config in phc/data/cfg/env
4. add the robot names in whatever u find h1 and g1 (can be done by searching through all files (vscode ide can do this) in the workspace), such as humanoid.py and humanoid_amp.py. Pid params, torque limits, default pos is added in humanoid.py.
5. retarget the dataset using the method in https://github.com/ZhengyiLuo/PHC/issues/77#issuecomment-2425226165. Btw we should fit the shape first, and then fit the motion.
6. I dont know (currently working on 5)

IlanElfen, 2024-11-05:
Thanks for your response! We'll see who can get to step 6 first :)

ZhengyiLuo, 2024-11-08:
The command for fitting AMASS motion to a new robot would be:

```

python scripts/data_process/fit_smpl_motion.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +fit_all=True
python scripts/data_process/fit_smpl_shape.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +fit_all=True
python scripts/vis/vis_q_mj.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +motion_name=[name_1, name_2, etc]

```

which should be run in that order. 

Yes, each robot needs its own config yaml file and specific motion file. I will try to write more detailed documentation on this.

ZhengyiLuo, 2024-11-08:
Extend_config is the configuration used for extending the robot arms/legs in order to add an end-effector to it. If your robot has hands/toes you do not need it. Untree H1 does not have a hand/head joint so I added them to match SMPL's rough kinematic structure.

dbdxnuliba, 2024-11-10:
@luoye2333 @ZhengyiLuo ，could you please show the full command to evalute g1/h1 after trained,many thanks

dbdxnuliba, 2024-11-10:
I meet err when run  python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True
(isaac) rob@rob:~/rl/PHC$ python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True
2024-11-10 13:27:46,325 - INFO - logger - logger initialized
2024-11-10 13:27:47,186 - DEBUG - Setting JobRuntime:name=UNKNOWN_NAME
2024-11-10 13:27:47,186 - DEBUG - Setting JobRuntime:name=fit_smpl_motion
Error executing job with overrides: ['robot=unitree_g1_fitting', '+fit_all=True']
Traceback (most recent call last):
  File "scripts/data_process/fit_smpl_motion.py", line 200, in main
    jobs= [jobs[i:i + chunk] for i in range(0, len(jobs), chunk)]
ValueError: range() arg 3 must not be zero

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

> The command for fitting AMASS motion to a new robot would be:
>
> ```
>
> python scripts/data_process/fit_smpl_motion.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +fit_all=True
> python [...]

luoye2333, 2024-11-10:
@dbdxnuliba
for g1 and h1, create new dataset is not needed. use g1_dance.pkl downloaded by download_data.sh is ok.
If you want to fit a new, you should first fit smpl betas and scale parameter
```
python scripts/data_process/fit_smpl_shape.py robot=unitree_g1_fitting +fit_all=True +vis=True
```
Then fit the motion on the fitted smpl parameter. This will take several hours. So i recommend use a small dataset to test first.
```
python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True
```
To visualize the fitted motion file
```
python scripts/vis/vis_q_mj.py robot=unitree_g1_fitting +motion_name=amass_all
```

IlanElfen, 2024-11-10:
@dbdxnuliba Look at [this issue](#92)

dbdxnuliba, 2024-11-11:
> If you want to fit a new, you should first fit smpl betas and scale parameter

thanks，I'll have a try

dbdxnuliba, 2024-11-11:
> @dbdxnuliba Look at [this issue](https://github.com/ZhengyiLuo/PHC/issues/92)

ok,thank you very much

dbdxnuliba, 2024-11-14:
@luoye2333，sorry to trouble you 
I executed the following on November 11, and by November 14, it still wasn't completed. I only put a small part of the dataset in. Why is this phenomenon occurring?

(isaac) rob@rob:~/rl/PHC$ python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True
2024-11-11 23:06:33,853 - INFO - logger - logger initialized
2024-11-11 23:06:34,753 - DEBUG - Setting JobRuntime:name=UNKNOWN_NAME
2024-11-11 23:06:34,753 - DEBUG - Setting JobRuntime:name=fit_smpl_motion
key_names: ['0-ACCAD_Female1General_c3d_A2 - Sway_poses', '0-ACCAD_Female1General_c3d_A2 - Sway t2_poses', '0-ACCAD_Female1General_c3d_A1 - Stand_poses']
enter process_motion
enter process_motion
enter process_motion
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 43/43 [00:00<00:00, 137.04it/s]
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 84/84 [00:00<00:00, 112347.43it/s]
load forward kinematics model
load smpl model
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 43/43 [00:00<00:00, 136.06it/s]
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 84/84 [00:00<00:00, 115628.99it/s]
load forward kinematics model
load smpl model
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 43/43 [00:00<00:00, 114.48it/s]
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 84/84 [00:00<00:00, 117558.07it/s]
load forward kinematics model
load smpl model

here is the dataset I put in only ACCAD folder， and only 3 npz files in it， 

```
rob@rob:~/rl/PHC/amass_data/Amass_part$ tree 
.
└── ACCAD
    └── Female1General_c3d
        ├── A1 - Stand_poses.npz
        ├── A2 - Sway_poses.npz
        └── A2 - Sway t2_poses.npz

2 directories, 3 files

here is the amas_root path
amass_root = "/home/rob/rl/PHC/amass_data/Amass_part"

hahahalove57, 2024-12-11:
> ，你好，很抱歉给你添麻烦我在 11 月 11 日执行了以下内容，到 11 月 14 日，它仍然没有完成。我只放入了数据集的一小部分。为什么会出现这种现象？
> 
> （isaac） rob@rob：~/rl/PHC$ python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True 2024-11-11 23：06：33,853 - INFO - logger - logger 初始化 2024-11-11 23：06：34,753 - DEBUG - 设置 JobRuntime：name=UNKNOWN_NAME 2024-11-11 23：06：34,753 - DEBUG - 设置 JobRuntime：name=fit_smpl_motionkey_names： ['0-ACCAD_Female1General_c3d_A2 - Sway_poses'， '0-ACCAD_Female1General_c3d_A2 - 摇摆t2_poses'， '0-ACCAD_Female1General_c3d_A1 - Stand_poses'] 输入 process_motion 输入 process_motion 输入 process_motion 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████|43/43 [00：00<00：00， 137.04it/s] 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████|84/84 [00：00<00：00， 112347.43it/s] 负载 正向运动学模型 负载 smpl 模型 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████|43/43 [00：00<00：00， 136.06it/s] 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████|84/84 [00：00<00：00， 115628.99it/s] 负载 正向运动学模型 负载 SMPL 模型 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████|43/43 [00：00<00：00， 114.48it/s] 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████|84/84 [00：00<00：00， 117558.07it/s] 负载 正向运动学模型 负载 SMPL 模型
> 
> 从添加的打印信息来看，似乎已知它卡在这个声明上 ![NAME](https://private-user-images.githubusercontent.com/13350158/386028661-5a487571-a0a5-49bd-a92e-7749eb1f8977.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5MjcwNjYsIm5iZiI6MTczMzkyNjc2NiwicGF0aCI6Ii8xMzM1MDE1OC8zODYwMjg2NjEtNWE0ODc1NzEtYTBhNS00OWJkLWE5MmUtNzc0OWViMWY4OTc3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjExVDE0MTkyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlNmQ1OWVmMDFmMzdlMTU2MzFlNTVkYTkyNjdhOTg0ZDdiMTBiNTY0NjJlODM5OTIyZDE1YzgzOGYxYmQ0ODImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.k5QzlfIuJZKkkyncVS4IVMiGNQ0uSuSjdt2LRQnrG6s)
> 
> ```
> here is the dataset I put in only ACCAD folder， and only 3 npz files in it， 
> 
> rob@rob:~/rl/PHC/amass_data/Amass_part$ tree 
> .
> └── ACCAD
>     └── Female1General_c3d
>         ├── A1 - Stand_poses.npz
>         ├── A2 - Sway_poses.npz
>         └── A2 - Sway t2_poses.npz
> 
> 2 directories, 3 files
> 
> here is the amas_root path
> amass_root = "/home/rob/rl/PHC/amass_data/Amass_part"
> ```
Hello! I have met the same problem, have you solved it yet?：）

IlanElfen, 2024-12-11:
@hahahalove57 
https://github.com/ZhengyiLuo/PHC/issues/99

hahahalove57, 2024-12-12:
> @hahahalove57 This issue was reopened here with a solution #99

thanks a lot!!!

legnAray, 2024-12-16:
> > Hi I'm also confused on how you retarget a different robot. I'm using [human2humanoid](https://github.com/LeCAR-Lab/human2humanoid), which uses PCH to retarget to a different robot. I'll admit I'm a little lost. Could you please walk me through the files you added, changed, etc. Thank you very much!
> 
> 1. add .urdf, .xml and /meshes to phc/data/assets/robot
> 2. add a yaml file like unitree_g1.yaml in phc/data/cfg/robot
> 3. add a env config in phc/data/cfg/env
> 4. add the robot names in whatever u find h1 and g1 (can be done by searching through all files (vscode ide can do this) in the workspace), such as humanoid.py and humanoid_amp.py. Pid params, torque limits, default pos is added in humanoid.py.
> 5. retarget the dataset using the method in [Questions Regarding H1 and G1 Support #77 (comment)](https://github.com/ZhengyiLuo/PHC/issues/77#issuecomment-2425226165). Btw we should fit the shape first, and then fit the motion.
> 6. I dont know (currently working on 5)

hey,  I want to use another G1 model file to replace the original one. (g1_29dof without hand&fingers)

I followed the steps you described：
> 1. add g1_29dof.xml and /meshes to phc/data/assets/robot
> 2. download the AMASS dataset I need and run fit_smpl_shape.py / fit_smpl_motion.py
> 3. run vis_q_mj.py , the result seems very OK
> 4. add a yaml file like unitree_g1.yaml in phc/data/cfg/robot
> 5. add a env config in phc/data/cfg/env
> 6. add g1_29dof in whatever I find h1 and g1.

I use this command to train:
`python phc/run_hydra.py project_name=g1_29dof_IM exp_name=g1_29dof_ACCAD_all robot=unitree_g1_29dof learning=im_pnn_big env=env_im_g1_29dof_phc env.motion_file=data/g1_29dof/v1/ACCAD.pkl sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7` 

but there is some problem :
[image]
my eps_len is convergenced at 2.7s.
It seems that there is a very strong external force that makes the robot "fly".

https://github.com/user-attachments/assets/bd157ba2-dfca-40a4-851f-8339592d0fbd

May I ask what problem this could be?

dbdxnuliba, 2024-12-21:
@legnAray  Hello,,I meet same issue ,have you solve the problem,could you provide an advice

QiuYi111, 2024-12-21:
@legnAray @ZhengyiLuo  Same issue here. even after trainning over 9000 iterations, eps_len converges to near 2.0 and the model performs poorly. I retargeted amass dataset to my own robot ,and followed the steps described before.

QiuYi111, 2024-12-23:
@legnAray  it worked! Thank you so much.

dbdxnuliba, 2025-01-01:
@legnAray ，Hello，
I have add <joint armature="1e-2"/> in xml,but the episode_length is still too low
[image]

and my xml is the following
```
<mujoco model="self_rob">
  <compiler angle="radian" meshdir="meshes/" autolimits="true" discardvisual="false"/>
  <statistic meansize="0.144785" extent="1.23314" center="0.025392 2.0634e-05 -0.245975"/>
  <default>
    **<joint armature="1e-2"/>**
    <default class="visual">
      <geom contype="0" conaffinity="0" group="0" density="0"/>
    </default>
    <default class="collision">
      <geom group="1"/>
    </default>
  </default>
    <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
        width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
    <texture name="texplane2" type="2d" builtin="checker" rgb1="1 0.3137 0.1843" rgb2="0.0 0.30196 0.38039"
        width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0." texture="texplane" texrepeat="1 1" texuniform="true"/>
    <material name="matplane2" reflectance="0.1" texture="texplane2" texrepeat="1 1" texuniform="true"/>

    <material name='obstacle'  rgba='0.9 0.6 0.2 1'/>
    <material name='visualgeom'  rgba='0.5 0.9 0.2 1'/>
    <material name='visualgeom2'  rgba='0.5 0.9 0.2 1'/>
    <mesh name="base_link" file="base_link.STL"/>
    <mesh name="base_link_smf" file="base_link_smf.STL"/>
    <mesh name="neck_yaw_link" file="neck_yaw_link.STL"/>
    <mesh name="neck_pitch_link" file="neck_pitch_link.STL"/>
    <mesh name="mic_link" file="mic_link.STL"/>
    <mesh name="head_rs_link" file="head_rs_link.STL"/>
    <mesh name="left_arm_base_link" file="left_arm_base_link.STL"/>
    <mesh name="left_shoulder_pitch_link" file="left_shoulder_pitch_link.STL"/>
    <mesh name="left_shoulder_roll_link" file="left_shoulder_roll_link.STL"/>
    <mesh name="left_arm_yaw_link" file="left_arm_yaw_link.STL"/>
    <mesh name="left_elbow_pitch_link" file="left_elbow_pitch_link.STL"/>
    <mesh name="left_elbow_yaw_link" file="left_elbow_yaw_link.STL"/>
    <mesh name="left_wrist_pitch_link" file="left_wrist_pitch_link.STL"/>
    <mesh name="left_wrist_pitch_link_smf" file="left_wrist_pitch_link_smf.STL"/>

    
    <mesh name="left_wrist_roll_link" file="left_wrist_roll_link.STL"/>
    <mesh name="left_hand_base_link" file="left_hand_base_link.STL"/>
    <mesh name="left_hand_ee_link" file="left_hand_ee_link.STL"/>
    <mesh name="right_arm_base_link" file="right_arm_base_link.STL"/>
    <mesh name="right_shoulder_pitch_link" file="right_shoulder_pitch_link.STL"/>
    <mesh name="right_shoulder_roll_link" file="right_shoulder_roll_link.STL"/>
    <mesh name="right_arm_yaw_link" file="right_arm_yaw_link.STL"/>
    <mesh name="right_elbow_pitch_link" file="right_elbow_pitch_link.STL"/>
    <mesh name="right_elbow_yaw_link" file="right_elbow_yaw_link.STL"/>
    <mesh name="right_wrist_pitch_link" file="right_wrist_pitch_link.STL"/>
    <mesh name="right_wrist_roll_link" file="right_wrist_roll_link.STL"/>
    <mesh name="right_hand_base_link" file="right_hand_base_link.STL"/>
    <mesh name="right_hand_ee_link" file="right_hand_ee_link.STL"/>
    <mesh name="waist_base_link" file="waist_base_link.STL"/>
    <mesh name="waist_pitch_link" file="waist_pitch_link.STL"/>
    <mesh name="waist_roll_link" file="waist_roll_link.STL"/>
    <mesh name="waist_roll_link_smf" file="waist_roll_link_smf.STL"/>

    <mesh name="waist_yaw_link" file="waist_yaw_link.STL"/>
    <mesh name="waist_yaw_link_smf" file="waist_yaw_link_smf.STL"/>

    <mesh name="left_hip_roll_link" file="left_hip_roll_link.STL"/>
    <mesh name="left_hip_yaw_link" file="left_hip_yaw_link.STL"/>
[... 336 lines cut ...]
    <motor name="waist_yaw_joint" joint="waist_yaw_joint"  ctrllimited="true" ctrlrange="-100 100"/>
    <motor name="waist_roll_joint" joint="waist_roll_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="waist_pitch_joint" joint="waist_pitch_joint"  ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_shoulder_pitch_joint" joint="left_shoulder_pitch_joint"  ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_shoulder_roll_joint" joint="left_shoulder_roll_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_arm_yaw_joint" joint="left_arm_yaw_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_elbow_pitch_joint" joint="left_elbow_pitch_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_elbow_yaw_joint" joint="left_elbow_yaw_joint"  ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="left_wrist_pitch_joint" joint="left_wrist_pitch_joint" ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="left_wrist_roll_joint" joint="left_wrist_roll_joint"  ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_shoulder_pitch_joint" joint="right_shoulder_pitch_joint"  ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="right_shoulder_roll_joint" joint="right_shoulder_roll_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="right_arm_yaw_joint" joint="right_arm_yaw_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="right_elbow_pitch_joint" joint="right_elbow_pitch_joint"  ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_elbow_yaw_joint" joint="right_elbow_yaw_joint" ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_wrist_pitch_joint" joint="right_wrist_pitch_joint" ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_wrist_roll_joint" joint="right_wrist_roll_joint" ctrllimited="true" ctrlrange="-35 35"/>
  </actuator>

</mujoco>

```

dou you have some advices ,and could you please provide your xml and some advices,
many thanks