ZhengyiLuo/PHC #86: Queations about imitation inference

PeterWangyi, 2024-11-06:
Hello author, thank you for your excellent work.

I have some questions about the inference stage. In some of my previous experiences, such as Deepmimic and AMP methods, a separate model is trained for each motion. During inference, only need to specify the init state, and the rest is left to the model itself. In PHC, the model learns a lot of motions. In my understanding, the inference stage should only specify the first frame action, and let the model infer the rest.

After I ran
python phc/run_hydra.py learning=im_mcp_big exp_name=phc_comp_3 env=env_im_getup_mcp robot=smpl_humanoid env.zero_out_far=False robot.real_weight_porpotion_boxes=False env.num_prim=3 env.motion_file=sample_data/amass_isaac_standing_upright_slim.pkl env.models=['output/HumanoidIm/phc_3/Humanoid.pth'] env.num_envs=1 headless=False epoch=-1 test=True
, and I replaced it with my own motion file.

In the following video, the input action is a walking motion with sliding. I observed that the model inference tends to fit the input motion frame by frame, rather than being able to correct the sliding problem.

So I am curious about how inference is done in PHC, and why it can imitate such a motion frame by frame?

Looking forward to your reply!

https://github.com/user-attachments/assets/f76a2c12-dc56-49b2-97b4-e5aeabcc1362

dbdxnuliba, 2024-11-08:
@PeterWangyi @ZhengyiLuo 
when I run the command, the agent motion is unnormal, here is the gif 
[image]
could you please help me  to fix it

ZhengyiLuo, 2024-11-08:
Hi! Thanks for your interest in my work. 

PHC is a motion-tracking policy that imitates per-frame reference motion. This formulation allows it to imitate many different types of motion (unlike Deepmimic/AMP, which is tethered to one type of motion). The reward for training PHC is to get as close to the reference motion as possible. 

As a result, PHC does not really try to "fix" foot sliding as its primary goal is to "imitate" the reference motion. It will try to imitate the foot sliding motion as much as it can using physically plausible actions. The foot jittering and high-frequency motion you are observing is PHC trying to achieve balance while fitting the reference motion that has the foot sliding. 

As for your second video it looks like the agent is imitating the reference motion decently? The reference (red dots) is just standing still without moving.

PeterWangyi, 2024-11-08:
@dbdxnuliba 
This is normal. There is env=env_im_getup_mcp in the command. During initialization, the character will fall down in the first round, and the reference state init will be executed in the second round of initialization. So it falls down first and then stands up.

You can try pressing the "M" key and then "J" in the window, which is also described in the readme, and you can see the motion of character getup.

PeterWangyi, 2024-11-08:
> PHC is a motion-tracking policy that imitates per-frame reference motion. This formulation a

@ZhengyiLuo 
Thanks for your reply!

I also realized this problem: the physics-based frame-by-frame imitate method does not seem to be suitable for use as a post-processing module to optimize non-physical phenomena in motion.

I read some other papers, such as PhysDiff, which added physics modules to the training process of motion generation, but the efficiency was very low. In addition, do you have any other suggestions?

Looking forward to your insights and replies.

kexul, 2024-11-08:
@PeterWangyi [Michael Black](https://www.youtube.com/@MichaelBlackMPI) had a paper about that. https://www.youtube.com/watch?v=Dufvp_O0ziU

dbdxnuliba, 2024-11-08:
@PeterWangyi ,thanks for your answer ,and may I ask you another quesition about how can you get your motion file , and what's the full command  do you use like  **abc.pkl**

python phc/run_hydra.py learning=im_mcp_big exp_name=phc_comp_3 env=env_im_getup_mcp robot=smpl_humanoid env.zero_out_far=False robot.real_weight_porpotion_boxes=False env.num_prim=3 env.motion_file=sample_data/**abc.pkl** env.models=['output/HumanoidIm/phc_3/Humanoid.pth'] env.num_envs=1 headless=False epoch=-1 test=True

and could you please send me your comand and motion file 
Many thanks
> Hello author, thank you for your excellent work.
> 
> I have some questions about the inference stage. In some of my previous experiences, such as Deepmimic and AMP methods, a separate model is trained for each motion. During inference, only need to specify the init state, and the rest is left to the model itself. In PHC, the model learns a lot of motions. In my understanding, the inference stage should only specify the first frame action, and let the model infer the rest.
> 
> After I ran python phc/run_hydra.py learning=im_mcp_big exp_name=phc_comp_3 env=env_im_getup_mcp robot=smpl_humanoid env.zero_out_far=False robot.real_weight_porpotion_boxes=False env.num_prim=3 env.motion_file=sample_data/amass_isaac_standing_upright_slim.pkl env.models=['output/HumanoidIm/phc_3/Humanoid.pth'] env.num_envs=1 headless=False epoch=-1 test=True , and I replaced it with my own motion file.
> 
> In the following video, the input action is a walking motion with sliding. I observed that the model inference tends to fit the input motion frame by frame, rather than being able to correct the sliding problem.
> 
> So I am curious about how inference is done in PHC, and why it can imitate such a motion frame by frame?
> 
> Looking forward to your reply!
> 
>  phc_comp_kp_2-2024-11-06-14.05.53.mp4

ZhengyiLuo, 2024-12-10:
Hi! @dbdxnuliba. Please follow the [dataset processing](https://github.com/ZhengyiLuo/PHC?tab=readme-ov-file#data-processing-amass) steps to download and process the AMASS dataset. Thanks!

closed 2024-12-10 by ZhengyiLuo