ZhengyiLuo/PHC #89: could you please give us a command to evluate go1 and h1 trained result

dbdxnuliba, 2024-11-08:
could you please give us a command to evluate go1 and h1 trained result

the train command 
```
python phc/run_hydra.py project_name=Robot_IM robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/dance_sample_g1.pkl learning=im_pnn_big exp_name=unitree_g1_pnn sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7 
```
but what about the evalutation command

ZhengyiLuo, 2024-11-08:
Hi! H1 and G1 do not yet have a released model. Would that be of interest?

dbdxnuliba, 2024-11-08:
> Hi! H1 and G1 do not yet have a released model. Would that be of interest?

Yes ,I'm very interested in indeed

ghy0626-design, 2024-11-09:
Hi! I used the following command to train the G1 robot model:

python phc/run_hydra.py project_name=Robot_IM robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/dance_sample_g1.pkl learning=im_pnn_big exp_name=unitree_g1_pnn sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7

However, I’m not sure how to evaluate it. Could you provide me with a command or guidance on how to evaluate the trained model? Thanks!

ghy0626-design, 2024-11-09:
> Hi! H1 and G1 do not yet have a released model. Would that be of interest?

Hi! I used the following command to train the G1 robot model:

python phc/run_hydra.py project_name=Robot_IM robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/dance_sample_g1.pkl learning=im_pnn_big exp_name=unitree_g1_pnn sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7

However, I’m not sure how to evaluate it. Could you provide me with a command or guidance on how to evaluate the trained model? Thanks!

closed 2024-12-02 by dbdxnuliba

legnAray, 2024-12-03:
hi, do u know how to evaluate yet？