ZhengyiLuo/PHC #92: failed to run python scripts/data_process/fit_smpl_motion.py to produce retargetted motion file about g1/h1

dbdxnuliba, 2024-11-10:
Hello,thanks for your greate work to train g1/h1
When I try to learn produce retargetted motion file about g1/h1 following  https://github.com/ZhengyiLuo/PHC/issues/77#issuecomment-2425226165 , I was failed .

Here is the detail:
(isaac) rob@rob:~/rl/PHC$ python scripts/data_process/fit_smpl_motion.py robot=[unitree_g1_fitting,unitree_h1_fitting] +fit_all=True
2024-11-10 13:12:17,796 - INFO - logger - logger initialized
2024-11-10 13:12:18,715 - DEBUG - Setting JobRuntime:name=UNKNOWN_NAME
2024-11-10 13:12:18,715 - DEBUG - Setting JobRuntime:name=fit_smpl_motion
Error executing job with overrides: ['robot=[unitree_g1_fitting,unitree_h1_fitting]', '+fit_all=True']
Traceback (most recent call last):
  File "scripts/data_process/fit_smpl_motion.py", line 200, in main
    jobs= [jobs[i:i + chunk] for i in range(0, len(jobs), chunk)]
ValueError: range() arg 3 must not be zero

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace
and  another try the command,following  has also been failed
(isaac) rob @rob:~/rl/PHC$ python scripts/data_process/fit_smpl_motion.py robot=[unitree_g1_fitting, unitree_h1_fitting, etc] +fit_all=True
2024-11-10 13:14:47,051 - INFO - logger - logger initialized
2024-11-10 13:14:47,893 - DEBUG - Setting JobRuntime:name=UNKNOWN_NAME
2024-11-10 13:14:47,893 - DEBUG - Setting JobRuntime:name=fit_smpl_motion
no viable alternative at input '[unitree_g1_fitting,'
See https://hydra.cc/docs/1.2/advanced/override_grammar/basic for details

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

Could you please tell me the full command to retarget g1/h1, many thanks

IlanElfen, 2024-11-10:
The issue is that in line 188 of fit_smpl_motion.py, you didn't change the directory of the AMASS Dataset.

```amass_root = "/hdd/zen/data/ActBound/AMASS/AMASS_Complete"```,

so the number of chunks is 0 and the number of jobs becomes 0 (creating an error)
You have to change the amass_root to the directory of your unzipped AMASS dataset.

dbdxnuliba, 2024-11-10:
ok，I will have a try ,thanks

dbdxnuliba, 2024-11-10:
> The issue is that in line 188 of fit_smpl_motion.py, you didn't change the directory of the AMASS Dataset.
>
> `amass_root = "/hdd/zen/data/ActBound/AMASS/AMASS_Complete"`,
>
> so the number of chunks is [...]

ok，I will have a try ,thanks

closed 2024-11-28 by dbdxnuliba