ZhengyiLuo/PHC #96: Mesh Generated from fit_smpl_motion Differs From Original Mesh (Inaccurate Retargeting)

IlanElfen, 2024-11-11:
Hi
I've been trying to run the retargeting pipeline for my own robot. The results were consistently bad and I've figured out the issue stems from the mesh generated by fit_smpl_motion has completely wrong joint positions.
When loading my .xml file into the program (and making corresponding .yaml files), this mesh is generated

[image]

Whereas the same .xml file is output as this is Isaac Gym:
[image]

The right arm is missing and the left leg is skewed at a strange angle. I'm having trouble understanding the source of this issue.
Does anyone know what would cause the discrepancy? Thank you!

luoye2333, 2024-11-12:
fit_smpl_motion.py use shape_optimized_v1.pkl to retarget smpl dataset. Do u run fit_smpl_shape.py before it? Plz use +vis=True to see if the matched joints are fitted well. You can use my changed code (#95) to better visualize the results.

luoye2333, 2024-11-12:
It just uses "joint_matches" in the robot .yaml file to compute loss. Other bodies are not used. So this problem is not in shape mismatch. The result in your picture shows that hand is not fitted well.

IlanElfen, 2024-11-12:
How can I make it fit better? I updated the image to your changed code

IlanElfen, 2024-11-12:
Yes! It looks much better, the loss is around 40 whereas it was 200+ before. I'll try to tweak robot.yaml to lower it further. 
Thank you!
[image]

For the g_rot issue, the problem stems from the population of the positions_world and rotations_world (torch_humanoid_batch, line 235-246). J has length 65, but some indexes in rotations_world don't have an associated tensor. There are only 51 valid indexes, which is how I get the g_rot to body_names mismatch.

For example, `rotations_world[self._parents[0]]` looks like this: 
```
tensor([[[[[ 0.0862, -0.9963,  0.0000],
           [ 0.9963,  0.0862,  0.0000],
           [ 0.0000,  0.0000,  1.0000]]],
         [[[ 0.0685, -0.9977,  0.0000],
           [ 0.9977,  0.0685,  0.0000],
           [ 0.0000,  0.0000,  1.0000]]],
          ...
```
but at index 37, it has no tensor
```
tensor([], size=(1, 90, 0, 3, 3), grad_fn=<UnsafeViewBackward0>)
```
Giving it an index out of bounds when it tries to access `rotations_world[self._parents[i]][:, :, 0]`
I'm confused where these tensors come from, I don't see where they're appended or created. Any help is much appreciated!

IlanElfen, 2024-11-12:
But if I copy-paste the rotations_world and positions_world from fit_smpl_shape.py to use it in fit_smpl_motion.py, the mesh is good and I have no errors.
[image]
Is this a bad idea?

IlanElfen, 2024-11-13:
I figured out the error and am posting the solution here if anyone has the same problem:
My XML file had many more bodies than joints, and many bodies were completely missing joints (not only the end effectors). This caused the error mentioned above.
The fix it, I added a joint to the previous unjointed ones with barely any possible movement so it's essentially static (but still populates everything correctly). It shouldn't have, but it took me forever. Enjoy
Thanks again for this work and thank you @luoye2333  for helping me debug!

closed 2024-11-13 by IlanElfen