ZhengyiLuo/PHC #99: I executed fit_smpl_motion on November 11, and by November 14, it still wasn't completed. I only put a small part of the dataset in. Why is this phenomenon occurring?

dbdxnuliba, 2024-11-14:
I executed thefollowing on November 11, and by November 14, it still wasn't completed. I only put a small part of the dataset in. Why is this phenomenon occurring?

(isaac) rob@rob:~/rl/PHC$ python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True
2024-11-11 23:06:33,853 - INFO - logger - logger initialized
2024-11-11 23:06:34,753 - DEBUG - Setting JobRuntime:name=UNKNOWN_NAME
2024-11-11 23:06:34,753 - DEBUG - Setting JobRuntime:name=fit_smpl_motion
key_names: ['0-ACCAD_Female1General_c3d_A2 - Sway_poses', '0-ACCAD_Female1General_c3d_A2 - Sway t2_poses', '0-ACCAD_Female1General_c3d_A1 - Stand_poses']
enter process_motion
enter process_motion
enter process_motion
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 43/43 [00:00<00:00, 137.04it/s]
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 84/84 [00:00<00:00, 112347.43it/s]
load forward kinematics model
load smpl model
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 43/43 [00:00<00:00, 136.06it/s]
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 84/84 [00:00<00:00, 115628.99it/s]
load forward kinematics model
load smpl model
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 43/43 [00:00<00:00, 114.48it/s]
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 84/84 [00:00<00:00, 117558.07it/s]
load forward kinematics model
load smpl model

here is the dataset I put yin 

```
rob@rob:~/rl/PHC/amass_data/Amass_part$ tree 
.
└── ACCAD
    └── Female1General_c3d
        ├── A1 - Stand_poses.npz
        ├── A2 - Sway_poses.npz
        └── A2 - Sway t2_poses.npz

2 directories, 3 files

```
import glob
import os
import sys
import pdb
import os.path as osp
sys.path.append(os.getcwd())

from smpl_sim.utils import torch_utils
from smpl_sim.poselib.skeleton.skeleton3d import SkeletonTree, SkeletonMotion, SkeletonState
from scipy.spatial.transform import Rotation as sRot
import numpy as np
import torch
from smpl_sim.smpllib.smpl_parser import (
    SMPL_Parser,
    SMPLH_Parser,
    SMPLX_Parser, 
)

import joblib
import torch
import torch.nn.functional as F
import math
from smpl_sim.utils.pytorch3d_transforms import axis_angle_to_matrix
from torch.autograd import Variable
from tqdm import tqdm
from smpl_sim.smpllib.smpl_joint_names import SMPL_MUJOCO_NAMES, SMPL_BONE_ORDER_NAMES, SMPLH_BONE_ORDER_NAMES, SMPLH_MUJOCO_NAMES
from phc.utils.torch_humanoid_batch import Humanoid_Batch
from smpl_sim.utils.smoothing_utils import gaussian_kernel_1d, gaussian_filter_1d_batch
from easydict import EasyDict
import hydra
from omegaconf import DictConfig, OmegaConf

def load_amass_data(data_path):
[... 176 lines cut ...]
        try:
            pool = Pool(num_jobs)   # multi-processing
            all_data_list = pool.starmap(process_motion, job_args)
        except KeyboardInterrupt:
            pool.terminate()
            pool.join()
        all_data = {}
        for data_dict in all_data_list:
            all_data.update(data_dict)
    import ipdb; ipdb.set_trace()
    if len(all_data) == 1:
        data_key = list(all_data.keys())[0]
        os.makedirs(f"data/{cfg.robot.humanoid_type}/v1/singles", exist_ok=True)
        dumped_file = f"data/{cfg.robot.humanoid_type}/v1/singles/{data_key}.pkl"
        print(dumped_file)
        joblib.dump(all_data, dumped_file)
    else:
        os.makedirs(f"data/{cfg.robot.humanoid_type}/v1/", exist_ok=True)
        joblib.dump(all_data, f"data/{cfg.robot.humanoid_type}/v1/amass_all.pkl")
    

if __name__ == "__main__":
    main()

```

luoye2333, 2024-11-14:
3 motion clips should take less than 1 minute (several minutes at most). Do you remove the breakpoint in line 217?
```
import ipdb; ipdb.set_trace()
```

IlanElfen, 2024-11-14:
I had the same issue. It works when you turn off multiprocessing (then it takes around a minute).
If you set num_jobs to 1, that should work, or process each file individually:
```
for pickle in all_pkls:
       all_pkls = [pickle]
       ...
       [the rest of the code in def main()]
```
But turning off multiprocessing increases the time if I want to retarget many files. The entire 16k AMASS motion files would take 7 days.
@luoye2333 Did you find a way to keep multiprocessing without it freezing?

dbdxnuliba, 2024-11-18:
@luoye2333 @IlanElfen @ZhengyiLuo ，before run the command，if we run export OMP_NUM_THREADS=1
and then run the command ： python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True，
3 motion clips should take less than 1 minute (several minutes at most).

closed 2024-11-26 by dbdxnuliba