ZhengyiLuo/PHC #101: when training  unitree_g1 and  replace the defaut motion_file with 0-DanceDB_20120807_CliodelaVara_Clio_Haniotikos_C3D_poses.pkl ,  after produce Humanoid_00001500.pth，and then it occurs an torch.OutOfMemoryError: CUDA out of memory error

dbdxnuliba, 2024-11-26:
Hello,when training  unitree_g1 and  replace the defaut motion_file with 0-DanceDB_20120807_CliodelaVara_Clio_Haniotikos_C3D_poses.pkl ,  after produce Humanoid_00001500.pth，and then it occurs an torch.OutOfMemoryError: CUDA out of memory error, the following is the detail: 
```
(phc) rob@lenovo-ThinkStation-PX:~/PHC$ python phc/run_hydra.py project_name=Robot_IM robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/0-DanceDB_20120807_CliodelaVara_Clio_Haniotikos_C3D_poses.pkl learning=im_pnn_big exp_name=unitree_g1_DanceDB_20120807_CliodelaVara_Clio_Haniotikos_C3D_poses sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7 
Importing module 'gym_38' (/home/rob/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/rob/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 2.4.1+cu121
Device count 3
/home/rob/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/rob/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/rob/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
2024-11-25 16:58:34,556 - INFO - logger - logger initialized
MOVING MOTION DATA TO GPU, USING CACHE: False
MOVING MOTION DATA TO GPU, USING CACHE: False
MOVING MOTION DATA TO GPU, USING CACHE: False
Importing module 'rlgpu_38' (/home/rob/isaacgym/python/isaacgym/_bindings/linux-x86_64/rlgpu_38.so)
2024-11-25 16:58:36,527 - DEBUG - Popen(['git', 'version'], cwd=/home/rob/PHC, stdin=None, shell=False, universal_newlines=False)
2024-11-25 16:58:36,544 - DEBUG - Popen(['git', 'version'], cwd=/home/rob/PHC, stdin=None, shell=False, universal_newlines=False)
2024-11-25 16:58:36,566 - DEBUG - Trying paths: ['/home/rob/.docker/config.json', '/home/rob/.dockercfg']
2024-11-25 16:58:36,566 - DEBUG - No config file found
2024-11-25 16:58:36,673 - DEBUG - Setting JobRuntime:name=UNKNOWN_NAME
2024-11-25 16:58:36,673 - DEBUG - Setting JobRuntime:name=run_hydra
wandb: Using wandb-core as the SDK backend.  Please refer to https://wandb.me/wandb-core for more information.
wandb: Currently logged in as: nuliguangxiao-yang. Use `wandb login --relogin` to force relogin
wandb: Tracking run with wandb version 0.18.6
wandb: Run data is saved locally in /home/rob/PHC/wandb/run-20241125_165838-drujpxsn
wandb: Run `wandb offline` to turn off syncing.
wandb: Syncing run twilight-puddle-9
wandb: ⭐️ View project at https://wandb.ai/nuliguangxiao-yang/Robot_IM
wandb: 🚀 View run at https://wandb.ai/nuliguangxiao-yang/Robot_IM/runs/drujpxsn
wandb: WARNING Calling wandb.run.save without any arguments is deprecated.Changes to attributes are automatically persisted.
torch_deterministic: False
torch_deterministic: False
torch_deterministic: False
Setting seed: 0
Started to train
/home/rob/anaconda3/envs/phc/lib/python3.8/site-packages/torch/functional.py:513: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3609.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
JointSpec type free not yet supported!
Cylinder is not natively supported, tesellated mesh will be used
Cylinder is not natively supported, tesellated mesh will be used
/home/rob/isaacgym/python/isaacgym/torch_utils.py:16: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  return torch.tensor(x, dtype=dtype, device=device, requires_grad=requires_grad)
Humanoid Weights [32.238920867443085, 32.238920867443085, 32.238920867443085, 32.238920867443085, 32.238920867443085, 32.238920867443085, 32.238920867443085, 32.238920867443085, 32.238920867443085, 32.238920867443085]
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
Warning: DOF limits are the same
[... 149 lines cut ...]
*********************************************************************************

100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3072/3072 [09:49<00:00,  5.21it/s]
0it [00:00, ?it/s]█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3072/3072 [09:49<00:00,  5.17it/s]
100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 3072/3072 [00:01<00:00, 2683.80it/s]
Error executing job with overrides: ['project_name=Robot_IM', 'robot=unitree_g1', 'env=env_im_g1_phc', 'env.motion_file=sample_data/0-DanceDB_20120807_CliodelaVara_Clio_Haniotikos_C3D_poses.pkl', 'learning=im_pnn_big', 'exp_name=unitree_g1_DanceDB_20120807_CliodelaVara_Clio_Haniotikos_C3D_poses', 'sim=robot_sim', 'control=robot_control', 'learning.params.network.space.continuous.sigma_init.val=-1.7']
Traceback (most recent call last):
  File "phc/run_hydra.py", line 337, in main
    runner.run(cfg)
  File "/home/rob/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/torch_runner.py", line 139, in run
    self.run_train()
  File "/home/rob/anaconda3/envs/phc/lib/python3.8/site-packages/rl_games/torch_runner.py", line 125, in run_train
    agent.train()
  File "/home/rob/PHC/phc/learning/common_agent.py", line 165, in train
    eval_info = self.eval()
  File "/home/rob/PHC/phc/learning/im_amp.py", line 179, in eval
    humanoid_env.begin_seq_motion_samples()
  File "/home/rob/PHC/phc/env/tasks/humanoid_im.py", line 473, in begin_seq_motion_samples
    self._motion_lib.load_motions(skeleton_trees=self.skeleton_trees, gender_betas=self.humanoid_shapes.cpu(), limb_weights=self.humanoid_limb_and_weights.cpu(), random_sample=False, start_idx=self.start_idx)
  File "/home/rob/PHC/phc/utils/motion_lib_real.py", line 205, in load_motions
    self.gvs = torch.cat([m.global_velocity for m in motions], dim=0).float().to(self._device)
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 1.69 GiB. GPU 0 has a total capacity of 44.52 GiB of which 4.19 MiB is free. Including non-PyTorch memory, this process has 44.49 GiB memory in use. Of the allocated memory 35.96 GiB is allocated by PyTorch, and 1.53 GiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

YechenZhou, 2024-11-26:
I solved this problem when I modified the value of num_envs to 1024 in ~/PHC/phc/data/cfg/env/env_im_g1_phc.yaml file. Hope this helps. @dbdxnuliba

dbdxnuliba, 2024-11-27:
thank

> I solved this problem when I modified the value of num_envs to 1024 in ~/PHC/phc/data/cfg/env/env_im_g1_phc.yaml file. Hope this helps. @dbdxnuliba

thanks, I'll have a try

closed 2024-11-28 by dbdxnuliba