ZhengyiLuo/PHC #106: how to evaluate unitree_g1

legnAray, 2024-12-03:
Hi Luo, thanks for your work for unitree_g1 !

I try to use your commands in readme to train unitree_g1:
`python phc/run_hydra.py project_name=Robot_IM robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/dance_sample_g1.pkl learning=im_pnn_big exp_name=unitree_g1_pnn sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7 `

but when I finish my training, I meet a question when i try to evaluate the result with the following commands:
`python phc/run_hydra.py robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/dance_sample_g1.pkl learning=im_pnn_big exp_name=unitree_g1_pnn +env.models=['output/HumanoidIm/unitree_g1_pnn/Humanoid_00006000.pth'] env.num_envs=1 headless=False epoch=-1 test=True`

In the simulation environment, my robot seems to be "twitching" all the time，and the reward seems very low...

reward: 0.29767152667045593 steps: 2.0
reward: 0.29767152667045593 steps: 2.0
reward: 0.29767152667045593 steps: 2.0
reward: 0.29767152667045593 steps: 2.0
reward: 0.29767152667045593 steps: 2.0
reward: 0.29767152667045593 steps: 2.0
reward: 0.29767152667045593 steps: 2.0

legnAray, 2024-12-04:
python phc/run_hydra.py project_name=Robot_IM learning=im_pnn_big exp_name=unitree_g1_pnn robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/dance_sample_g1.pkl epoch=-1 env.num_envs=1 headless=False test=True sim=robot_sim control=robot_control

closed 2024-12-04 by legnAray

HonglinChu, 2025-02-26:
when add  "im_eval=True" to evaluate on dance_sample_g1.pkl ,  the error as follow:
PHC/phc/utils/motion_lib_base.py", line 145, in <lambda>
    data_list = {item[0]: item[1] for item in sorted(self._motion_data_load.items(), key=lambda entry: len(entry[1]['pose_quat_global']), reverse=True)}
KeyError: 'pose_quat_global'