ZhengyiLuo/PHC #111: problems about writing the export_policy_as_jit() function.

smilefish110, 2024-12-20:
Hello, sir!  Could you please show us how to export the model in jit format for deployment? I have problems while writing the export_policy_as_jit() function. Thank you very much! @ZhengyiLuo

smilefish110, 2024-12-20:
I just found out that network input information cannot be obtained in real scenarios. So, is there any good way to solve the sim2real problem?
[image]

ZhengyiLuo, 2024-12-26:
Hi! The policy trained in this repo is more suitable as a teacher policy for teacher-student distillation than direct deployment. For deployment on the real humanoid, please refer to the [H2O](https://github.com/LeCAR-Lab/human2humanoid) code base.

closed 2024-12-26 by ZhengyiLuo

smilefish110, 2024-12-26:
Thank you, sir!