ZhengyiLuo/PHC #112: How to Set Agent's Initial Location Based on MDM-Generated Sequence?

Rowing0914, 2024-12-20:
Hello,

I’m using MDM to generate motion sequences (keypoints) for your pretrained PHC to follow, but I’m unsure how to set the agent's initial location in the simulation based on the first frame of the generated sequence, as seen in the demo videos on your website. Could you point me to the part of the code where this is handled?

I understand we can get per-timestep keypoints through the MDM API [here](https://github.com/ZhengyiLuo/PHC/blob/master/phc/env/tasks/humanoid_im_mcp_demo.py#L253), but I’m not sure where the agent's initial location is set in the codebase.

For example, [here](https://github.com/ZhengyiLuo/PHC/blob/562b90f2d5e59d877d959e45fcea95ef14c06a80/phc/env/tasks/humanoid_amp.py#L508), we set the initial location to the first frame of a motion from the motion library, which seems unrelated to MDM-generated motion. Am I missing something?

Thanks!

ZhengyiLuo, 2024-12-26:
Hi! There are a couple of things you can do. 

You can initialize the humanoid to a standing position (I provided a standing checkpoint). If you use the released checkpoint for ICCV, which has failure state recovery capability, it can walk to the MDM key points and start imitating. You will need to disable the resetting. 

In addition to the above, you can also make the MDM keypoints spawn at the same positions as the humanoid (by applying an offset). I provide [MDM demos here](https://github.com/ZhengyiLuo/PHC/blob/master/docs/language_to_control_demo.md).

Rowing0914, 2024-12-26:
Thank you for the explanation! I’ve already taken a look at the MDM and managed to resolve this issue. I appreciate your help!

Best,

closed 2024-12-26 by Rowing0914