ZhengyiLuo/PHC #113: eps_len converges  to nearly 2.0

QiuYi111, 2024-12-21:
> > Hi I'm also confused on how you retarget a different robot. I'm using [human2humanoid](https://github.com/LeCAR-Lab/human2humanoid), which uses PCH to retarget to a different robot. I'll admit I'm a little lost. Could you please walk me through the files you added, changed, etc. Thank you very much!
> 
> 1. add .urdf, .xml and /meshes to phc/data/assets/robot
> 2. add a yaml file like unitree_g1.yaml in phc/data/cfg/robot
> 3. add a env config in phc/data/cfg/env
> 4. add the robot names in whatever u find h1 and g1 (can be done by searching through all files (vscode ide can do this) in the workspace), such as humanoid.py and humanoid_amp.py. Pid params, torque limits, default pos is added in humanoid.py.
> 5. retarget the dataset using the method in [Questions Regarding H1 and G1 Support #77 (comment)](https://github.com/ZhengyiLuo/PHC/issues/77#issuecomment-2425226165). Btw we should fit the shape first, and then fit the motion.
> 6. I dont know (currently working on 5)

hey,  I want to use another G1 model file to replace the original one. (g1_29dof without hand&fingers)

I followed the steps you described：
> 1. add g1_29dof.xml and /meshes to phc/data/assets/robot
> 2. download the AMASS dataset I need and run fit_smpl_shape.py / fit_smpl_motion.py
> 3. run vis_q_mj.py , the result seems very OK
> 4. add a yaml file like unitree_g1.yaml in phc/data/cfg/robot
> 5. add a env config in phc/data/cfg/env
> 6. add g1_29dof in whatever I find h1 and g1.

I use this command to train:
`python phc/run_hydra.py project_name=g1_29dof_IM exp_name=g1_29dof_ACCAD_all robot=unitree_g1_29dof learning=im_pnn_big env=env_im_g1_29dof_phc env.motion_file=data/g1_29dof/v1/ACCAD.pkl sim=robot_sim control=robot_control learning.params.network.space.continuous.sigma_init.val=-1.7` 

but there is some problem :
[image]
my eps_len is convergenced at 2.7s.
It seems that there is a very strong external force that makes the robot "fly".

https://github.com/user-attachments/assets/bd157ba2-dfca-40a4-851f-8339592d0fbd

May I ask what problem this could be?

_Originally posted by @legnAray in https://github.com/ZhengyiLuo/PHC/issues/84#issuecomment-2545117296_
            
 Hello,Luo!  I have same issue here. even after trainning over 9000 iterations, eps_len converges to near 2.0 and the model performs poorly(same as @legnAray described). I retargeted amass dataset to my own robot ,and followed the steps described before. I opened this new issue in case that being burried by issues.

QiuYi111, 2024-12-23:
> @dbdxnuliba @QiuYi111 I added the following code to the .xml file, and now it seems to be working quite well.
> 
> ```
> <default>
>     <joint armature="1e-2"/>
> </default>
> ```

by @legnAray

closed 2024-12-23 by QiuYi111

zengweishuai, 2024-12-29:
@QiuYi111
Hi, I've encountered the same problem. Even if I modify the xml file as you mention, the reward and episode length do not increase either. I wonder what the normal curve would be like? Is it rising fast at the very beginning?

QiuYi111, 2024-12-29:
@zengweishuai Mostly it should looks like this 
[image]

I have encountered same problem occationaly. I find that if you do:
`    reward=100*reward
    reward_raw=100*reward_raw
`
in humanoid_im.py line 1553, the problem no longer happens.

zengweishuai, 2024-12-29:
> @zengweishuai Mostly it should looks like this [...]

@QiuYi111  Thanks for your quick reply! The curve is quite good! I wonder if there is anything wrong with my process:

1. I first retarget the AMASS dataset. However, the code given could not directly fit g1_29dof, so I change the joint matches and remove the hand matching and replace it with the wrist matching.
2. The env config in PHC should also be changed. I change the key_bodies and reset_bodies items in the config file to remove those that do not exist for g1_29dof.
3. The robot config in PHC should also be changed. I change the body_names, limb_weight_group and dof_names in it. The list of body_names in the end has 30 elements in all and the dof_names has 29 elements in all.
4. The whole training pipeline could not run until now. I also change the filter_ints, p_gains and d_gains to fit the dimension.

I wonder if you have done the same things to ensure successful running. If so, i wonder how you handle each details as for me, it can not work until now.

I appreciate it for your immediate assistance. Expect your reply!

dbdxnuliba, 2024-12-30:
@QiuYi111 ,hello ,
how can we set appropriate value for filter_ints 
in humanoid.py 
```

        if self.humanoid_type in ['h1', 'g1', 'self_rob',"smpl", "smplh", "smplx"] and self._has_self_collision:
            # compliance_vals = [0.1] * 24
            # thickness_vals = [1.0] * 24
            if self._has_mesh:
                filter_ints = [0, 1, 224, 512, 384, 1, 1792, 64, 1056, 4096, 6, 6168, 0, 2048, 0, 20, 0, 0, 0, 0, 10, 0, 0, 0]
            else:
                if self.humanoid_type == "smpl":
                    filter_ints = [0, 0, 7, 16, 12, 0, 56, 2, 33, 128, 0, 192, 0, 64, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
                elif self.humanoid_type in ["smplh", "smplx"]:
                    filter_ints = [0, 0, 7, 16, 12, 0, 56, 2, 33, 128, 0, 192, 0, 64, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
                elif self.humanoid_type in ['h1']:
                    filter_ints = [0, 2, 0, 2, 0, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
                elif self.humanoid_type in ['g1']:
                    filter_ints = [0, 0, 2688, 0, 8192, 0, 8192, 0, 1344, 0, 4096, 0, 4096, 3072, 768, 192, 1, 0, 1, 0, 32, 8, 40, 0, 0, 0, 0, 0, 2, 0, 2, 0, 16, 4, 20, 0, 0, 0, 0, 0]
                elif self.humanoid_type in ['self_rob']:
                    print('self_rob set filter_ints')
                    filter_ints = [0,
                                    300, 2688, 300, 8192, 0, 8192,
                                    1, 1344, 1, 4096, 0, 4096, 
                                    3072, 4072, 5072,
                                    0, 3, 0, 3, 0, 3, 0,
                                    0,2, 0, 2, 0, 2, 0]
                    
            props = self.gym.get_actor_rigid_shape_properties(env_ptr, humanoid_handle)
            print('len(filter_ints):', len(filter_ints))
            print('len(props):', len(props))
            assert (len(filter_ints) == len(props))

            for p_idx in range(len(props)):
                props[p_idx].filter = filter_ints[p_idx]
            self.gym.set_actor_rigid_shape_properties(env_ptr, humanoid_handle, props)
```

QiuYi111, 2024-12-30:
@zengweishuai I suggest that you see this issue [84](https://github.com/ZhengyiLuo/PHC/issues/84) to properly add your own robot. I followed and it works well.

QiuYi111, 2024-12-30:
@dbdxnuliba I actually first fit the dimension and then I configured to avoid collision between links on the same limb. For example hand and arm do not have collision.  But to be honest, I did not spend time to figure out a better logic.

dbdxnuliba, 2025-01-01:
@QiuYi111  ，Hello，
I have add <joint armature="1e-2"/> in xml,but the episode_length is still too low
[image]
   
**when train :**
self_rob.yaml
set has_self_collision: True
[image]

**when evaluation**
and after 18000 iterations, when evaluate ,it shows as the following:

[image]

So 

**Why is the joint movement of this agent not continuous during training? and why the trained result not work?Is it because of the collision settings in my XML or the corresponding filter_ints not being set correctly?**
**and self_rob.xml define** 

```
<mujoco model="self_rob">
  <compiler angle="radian" meshdir="meshes/" autolimits="true" discardvisual="false"/>
  <statistic meansize="0.144785" extent="1.23314" center="0.025392 2.0634e-05 -0.245975"/>
  <default>
    <joint armature="1e-2"/>
    <default class="visual">
      <geom contype="0" conaffinity="0" group="0" density="0"/>
    </default>
    <default class="collision">
      <geom group="1"/>
    </default>
  </default>
    <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
        width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
    <texture name="texplane2" type="2d" builtin="checker" rgb1="1 0.3137 0.1843" rgb2="0.0 0.30196 0.38039"
        width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0." texture="texplane" texrepeat="1 1" texuniform="true"/>
    <material name="matplane2" reflectance="0.1" texture="texplane2" texrepeat="1 1" texuniform="true"/>

    <material name='obstacle'  rgba='0.9 0.6 0.2 1'/>
    <material name='visualgeom'  rgba='0.5 0.9 0.2 1'/>
    <material name='visualgeom2'  rgba='0.5 0.9 0.2 1'/>
    <mesh name="base_link" file="base_link.STL"/>
    <mesh name="base_link_smf" file="base_link_smf.STL"/>
    <mesh name="neck_yaw_link" file="neck_yaw_link.STL"/>
    <mesh name="neck_pitch_link" file="neck_pitch_link.STL"/>
    <mesh name="mic_link" file="mic_link.STL"/>
    <mesh name="head_rs_link" file="head_rs_link.STL"/>
    <mesh name="left_arm_base_link" file="left_arm_base_link.STL"/>
    <mesh name="left_shoulder_pitch_link" file="left_shoulder_pitch_link.STL"/>
    <mesh name="left_shoulder_roll_link" file="left_shoulder_roll_link.STL"/>
    <mesh name="left_arm_yaw_link" file="left_arm_yaw_link.STL"/>
    <mesh name="left_elbow_pitch_link" file="left_elbow_pitch_link.STL"/>
    <mesh name="left_elbow_yaw_link" file="left_elbow_yaw_link.STL"/>
    <mesh name="left_wrist_pitch_link" file="left_wrist_pitch_link.STL"/>
    <mesh name="left_wrist_pitch_link_smf" file="left_wrist_pitch_link_smf.STL"/>

    
    <mesh name="left_wrist_roll_link" file="left_wrist_roll_link.STL"/>
    <mesh name="left_hand_base_link" file="left_hand_base_link.STL"/>
    <mesh name="left_hand_ee_link" file="left_hand_ee_link.STL"/>
    <mesh name="right_arm_base_link" file="right_arm_base_link.STL"/>
    <mesh name="right_shoulder_pitch_link" file="right_shoulder_pitch_link.STL"/>
    <mesh name="right_shoulder_roll_link" file="right_shoulder_roll_link.STL"/>
    <mesh name="right_arm_yaw_link" file="right_arm_yaw_link.STL"/>
    <mesh name="right_elbow_pitch_link" file="right_elbow_pitch_link.STL"/>
    <mesh name="right_elbow_yaw_link" file="right_elbow_yaw_link.STL"/>
[... 347 lines cut ...]
    <motor name="right_knee_joint" joint="right_knee_joint"  ctrllimited="true" ctrlrange="-360.99 360.99"/>
    <motor name="right_ankle_pitch_joint" joint="right_ankle_pitch_joint"  ctrllimited="true" ctrlrange="-43.888 43.888"/>
    <motor name="right_ankle_roll_joint" joint="right_ankle_roll_joint" ctrllimited="true" ctrlrange="-43.888 43.888"/>
    <motor name="waist_yaw_joint" joint="waist_yaw_joint"  ctrllimited="true" ctrlrange="-100 100"/>
    <motor name="waist_roll_joint" joint="waist_roll_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="waist_pitch_joint" joint="waist_pitch_joint"  ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_shoulder_pitch_joint" joint="left_shoulder_pitch_joint"  ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_shoulder_roll_joint" joint="left_shoulder_roll_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_arm_yaw_joint" joint="left_arm_yaw_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_elbow_pitch_joint" joint="left_elbow_pitch_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="left_elbow_yaw_joint" joint="left_elbow_yaw_joint"  ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="left_wrist_pitch_joint" joint="left_wrist_pitch_joint" ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="left_wrist_roll_joint" joint="left_wrist_roll_joint"  ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_shoulder_pitch_joint" joint="right_shoulder_pitch_joint"  ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="right_shoulder_roll_joint" joint="right_shoulder_roll_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="right_arm_yaw_joint" joint="right_arm_yaw_joint" ctrllimited="true" ctrlrange="-110 110"/>
    <motor name="right_elbow_pitch_joint" joint="right_elbow_pitch_joint"  ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_elbow_yaw_joint" joint="right_elbow_yaw_joint" ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_wrist_pitch_joint" joint="right_wrist_pitch_joint" ctrllimited="true" ctrlrange="-35 35"/>
    <motor name="right_wrist_roll_joint" joint="right_wrist_roll_joint" ctrllimited="true" ctrlrange="-35 35"/>
  </actuator>

</mujoco>

```

QiuYi111, 2025-01-01:
> @zengweishuai Mostly it should looks like this 
> [image]
> 
> I have encountered same problem occationaly. I find that if you do:
> `    reward=100*reward
>     reward_raw=100*reward_raw
> `
> in humanoid_im.py line 1553, the problem no longer happens.

@dbdxnuliba  I think you should try do some scale on the reward. It works for me but I do not really understand why. Also you may clear cache first.

dbdxnuliba, 2025-01-01:
@QiuYi111 many thanks ,but how can we clear cache in the code,