ZhengyiLuo/PHC #114: what's the meanning of  filter_ints about collision and how could we set  appropriate value about filter_ints for self define robot type,

dbdxnuliba, 2024-12-30:
what's the meanning of  filter_ints about collision and how could we set  appropriate value about filter_ints for self define robot type,   
     if self.humanoid_type in ['h1', 'g1', 'self_rob',"smpl", "smplh", "smplx"] and self._has_self_collision:
            # compliance_vals = [0.1] * 24
            # thickness_vals = [1.0] * 24
            if self._has_mesh:
                filter_ints = [0, 1, 224, 512, 384, 1, 1792, 64, 1056, 4096, 6, 6168, 0, 2048, 0, 20, 0, 0, 0, 0, 10, 0, 0, 0]
            else:
                if self.humanoid_type == "smpl":
                    filter_ints = [0, 0, 7, 16, 12, 0, 56, 2, 33, 128, 0, 192, 0, 64, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
                elif self.humanoid_type in ["smplh", "smplx"]:
                    filter_ints = [0, 0, 7, 16, 12, 0, 56, 2, 33, 128, 0, 192, 0, 64, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
                elif self.humanoid_type in ['h1']:
                    **filter_ints = [0, 2, 0, 2, 0, 0, 1, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]**
                elif self.humanoid_type in ['g1']:
                    filter_ints = [0, 0, 2688, 0, 8192, 0, 8192, 0, 1344, 0, 4096, 0, 4096, 3072, 768, 192, 1, 0, 1, 0, 32, 8, 40, 0, 0, 0, 0, 0, 2, 0, 2, 0, 16, 4, 20, 0, 0, 0, 0, 0]
                elif self.humanoid_type in ['self_rob']:
                    **filter_ints = []**
            props = self.gym.get_actor_rigid_shape_properties(env_ptr, humanoid_handle)
            assert (len(filter_ints) == len(props))

            for p_idx in range(len(props)):
                props[p_idx].filter = filter_ints[p_idx]
            **self.gym.set_actor_rigid_shape_properties(env_ptr, humanoid_handle, props)**

HonglinChu, 2025-01-21:
I have the same question about “filter_int” setting for myself robot

ZhengyiLuo, 2025-02-08:
The filter_int is set based on the self-collision API from IsaacGym. IsaacGym does not read the collision setting from XML so I have to set them manually. It's a bit-wise binary setting where if a body part has the same bit set to 1, they no longer collide.

dbdxnuliba, 2025-02-14:
> The filter_int is set based on the self-collision API from IsaacGym. IsaacGym does not read the collision setting from XML so I have to set them manually. It's a bit-wise binary setting where if a body part has the same bit set to 1, they no longer collide.
Get,many thanks！

closed 2025-02-14 by dbdxnuliba