ZhengyiLuo/PHC #115: Input to Physics Engine

Jatinkalal, 2024-12-30:
Hi, I I have a doubt regarding input 3d keypoints to the physics engine, In UHC the humanoid is trained on AMASS poses which are axis angle representation, in which representation is PHC humanoid trained on?, because phc mimics the 3d keypoints from MLD, so is there any conversion you do ?
Thank you.

ZhengyiLuo, 2025-02-15:
Do you mean the input to PHC model? I use the 6D rotation representation. 3D keypoints I change the order but it's the same 24 keypoints.

Jatinkalal, 2025-02-18:
So you use the 263-D output from MDM (motion diffusion model) as input to PHC?
If so how can I use it for UHC? (as there is no trans parameter in 263-D)
Thank You.