ZhengyiLuo/PHC #116: The fitting problem of converting the ACCAD dataset to G1 data

XiaoWZENG, 2025-01-08:
Thank you for your work. I have encountered some issues during the reproduction process.
How to correctly convert the AMASS dataset into G1 data. The conversion process is stuck at the following prompt and does not continue to generate the pkl model.
████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 43/43 [00:01<00:00, 21.87it/s]
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 84/84 [00:00<00:00, 49727.81it/s]

Thanks a lot！

XiaoWZENG, 2025-01-08:
When I run the bellow command:
python scripts/data_process/fit_smpl_motion.py robot=unitree_g1_fitting +fit_all=True +amass_root=/home/aeon_jd/zzc_ws/new_work/humanoid/PHC-master/AMASS_DATA

sfy666, 2025-05-09:
hello，have you solved the problem？I met the same trouble