ZhengyiLuo/PHC #118: Imitation sticks to the first frame

avegetablechicken, 2025-01-16:
I try to evaluate on Human3.6M but it seems that the agent keeps trying to imitate the pose in the first frame. Should I change the `progress_buf` variable manually to make it go on for successive frames?

ZhengyiLuo, 2025-02-15:
Could you elaborate on the issue? Does the reference red dots changes but the humanoid not change?

avegetablechicken, 2025-02-25:
> Could you elaborate on the issue? Does the reference red dots changes but the humanoid does not move?

Sorry for being late. Due to server issues I had been unable to view the rendering window. Now I fix this problem. The video below shows the humanoid trying to track a motion clip.

https://github.com/user-attachments/assets/f4c9aeff-3e34-4841-b481-304245f8d818

Here is the console output when I add a line of code to print the `self.progress_buf` of class `Humanoid`:

[image]
It seems that even after several hours of training, the target frame is still the first frame. 

So the real question is how can I watch the process of slowly getting to track a predefined motion clip? (Now I find that the policy keeps tracking the first one or two frames.)