ZhengyiLuo/PHC #123: Question about H1/G1 robot success rate

HonglinChu, 2025-02-26:
I can success run the command as follow:
`python phc/run_hydra.py project_name=Robot_IM learning=im_pnn_big exp_name=unitree_g1_pnn robot=unitree_g1 env=env_im_g1_phc env.motion_file=sample_data/dance_sample_g1.pkl epoch=-1 env.num_envs=1 headless=False test=True sim=robot_sim control=robot_control
`
But,when add "im_eval=True" to calculate success rate, i got the error:
`
PHC/phc/utils/motion_lib_base.py", line 145, in <lambda>
    data_list = {item[0]: item[1] for item in sorted(self._motion_data_load.items(), key=lambda entry: len(entry[1]['pose_quat_global']), reverse=True)}
KeyError: 'pose_quat_global'
`

HonglinChu, 2025-02-27:
I solved the problem:
`
import joblib
from scipy.spatial.transform import Rotation as sRot
data = joblib.load('sample_data/dance_sample_g1.pkl')
for k,v in data.items():
    pose_aa = v['pose_aa'] # T,N*3
    T= pose_aa.shape[0] # 
    pose_aa_mj = pose_aa.reshape(T, -1, 3)
    pose_quat = sRot.from_rotvec(pose_aa_mj.reshape(-1, 3)).as_quat().reshape(T, -1, 4)
    data[k]['pose_quat_global'] = pose_quat 
joblib.dump(data, 'sample_data/dance_sample_g1_new.pkl')
`

closed 2025-02-27 by HonglinChu

WesleyXu224, 2025-08-03:
您好呀！想问问你是如何解决在PHC上使用G1/H1推理出指标的吗？