ZhengyiLuo/PHC #128: Xlib:  extension "GLX" missing on display ":1".

Zijing5, 2025-03-27:
**When I try to run the following command, I get an error as described in the title(Xlib:  extension "GLX" missing on display ":1".)：**
`python phc/run_hydra.py learning=im_pnn_big exp_name=phc_x_pnn  env=env_im_x_pnn robot=smplx_humanoid env.motion_file=sample_data/standing_x.pkl env.training_prim=0 epoch=-1 test=True  env.num_envs=1  headless=False`

**Many schemes on the Internet told me to check:**
```
grep -i glx /var/log/Xorg.0.log
[  6130.866] (II) LoadModule: "glx"
[  6130.867] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[  6130.867] (II) Module glx: vendor="X.Org Foundation"
[  6132.077] (II) Initializing extension GLX
[  6132.077] (II) AIGLX: Screen 0 is not DRI2 capable
[  6132.097] (II) IGLX: Loaded and initialized swrast
[  6132.097] (II) GLX: Initialized DRISWRAST GL provider for screen 0
```

**But I still don't quite know how to solve this problem.
Here is the full output of the terminal:**

```
Importing module 'gym_38' (/history_data/wzj/wzj_file/IG/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /history_data/wzj/wzj_file/IG/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 2.4.1+cu121
Device count 1
/history_data/wzj/wzj_file/IG/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/wzj/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/wzj/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
2025-03-27 18:42:24,415 - INFO - logger - logger initialized
MOVING MOTION DATA TO GPU, USING CACHE: False
MOVING MOTION DATA TO GPU, USING CACHE: False
Importing module 'rlgpu_38' (/history_data/wzj/wzj_file/IG/isaacgym/python/isaacgym/_bindings/linux-x86_64/rlgpu_38.so)
torch_deterministic: False
torch_deterministic: False
torch_deterministic: False
Setting seed: 0
Found checkpoint
output/HumanoidIm/phc_x_pnn/Humanoid.pth
Started to play
/home/wzj/.conda/envs/phc0/lib/python3.8/site-packages/torch/functional.py:513: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at ../aten/src/ATen/native/TensorShape.cpp:3609.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
Xlib:  extension "GLX" missing on display ":1".
```
**Who can help me? Thank you very much!**

ZhengyiLuo, 2025-04-08:
Try with `no_virtual_display=True`

Zijing5, 2025-04-08:
I have solved the problem! Thank you very much for your reply!

closed 2025-05-11 by ZhengyiLuo