ZhengyiLuo/PHC #136: why "only use the heading" in fit_smpl_motion.py

Philipflyg, 2025-04-24:
Why not use gt_root_rot_quat directly, but only use the heading?

>          gt_root_rot_quat = torch.from_numpy((sRot.from_rotvec(pose_aa_walk[:, :3]) * sRot.from_quat([0.5, 0.5, 0.5, 0.5]).inv()).as_quat()).float() # can't directly use this 
>          gt_root_rot = torch.from_numpy(sRot.from_quat(torch_utils.calc_heading_quat(gt_root_rot_quat)).as_rotvec()).float() # so only use the heading.

ZhengyiLuo, 2025-04-28:
Becuase the hit structure of the humanoid and the SMPL human is different, so directly using the root rotation of the SMPL human will cause problems for the humanoid.

Philipflyg, 2025-04-28:
> Becuase the hip structure of the humanoid and the SMPL human is different, so directly using the root rotation of the SMPL human will cause problems for the humanoid.

got it, thanks!

closed 2025-04-28 by Philipflyg