ZhengyiLuo/PHC #152: Crash when evaluation

onlyloveyanzi, 2025-08-11:
I encounter a crash every time I evaluate for about 7500 episodes, and sometimes the terminal even closes automatically. During the training process, the GPU memory usage is normal, but after the crash, the GPU memory is not automatically released. How can I resolve this issue?

Terminated: 360 | max frames: 3756 | steps 3755 | Start: 9216 | Succ rate: 0.721 | Mpjpe: 65.849:  90%|█████████████████████████████████████████████████▌     | 9/10 [1:30:55<10:29, 629.82s/it]
****************************** Current motion keys ******************************
Sampling motion: tensor([10240, 10241, 10242, 10243, 10244, 10245, 10246, 10247, 10248, 10249,
        10250, 10251, 10252, 10253, 10254, 10255, 10256, 10257, 10258, 10259,
        10260, 10261, 10262, 10263, 10264, 10265, 10266, 10267, 10268, 10269],
       device='cuda:0')
['0-CMU_CMU_39_39_14_poses' '0-CMU_CMU_39_39_12_poses' '0-CMU_CMU_39_39_06_poses' '0-CMU_CMU_39_39_09_poses' '0-CMU_CMU_39_39_10_poses' '0-CMU_CMU_39_39_04_poses' '0-CMU_CMU_39_39_01_poses' '0-CMU_CMU_39_39_11_poses' '0-CMU_CMU_39_39_13_poses' '0-CMU_CMU_39_39_03_poses' '0-CMU_CMU_39_39_05_poses' '0-CMU_CMU_02_02_07_poses' '0-CMU_CMU_02_02_05_poses' '0-CMU_CMU_02_02_02_poses' '0-CMU_CMU_02_02_01_poses' '0-CMU_CMU_02_02_09_poses' '0-CMU_CMU_02_02_06_poses' '0-CMU_CMU_02_02_04_poses' '0-CMU_CMU_02_02_10_poses' '0-CMU_CMU_02_02_08_poses' '0-CMU_CMU_02_02_03_poses' '0-CMU_CMU_133_133_07_poses' '0-CMU_CMU_133_133_15_poses' '0-CMU_CMU_133_133_14_poses' '0-CMU_CMU_133_133_05_poses' '0-CMU_CMU_133_133_19_poses' '0-CMU_CMU_133_133_17_poses' '0-CMU_CMU_133_133_26_poses' '0-CMU_CMU_133_133_04_poses' '0-CMU_CMU_133_133_03_poses'] .....
*********************************************************************************

                                                                                                                                                                                               Killed███████████████████████████████████████████████████████████████████████████████████████▌                                                                | 590/1024 [01:29<05:52,  1.23it/s]

and i run it on a single 4090  

Mon Aug 11 12:49:24 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.163.01             Driver Version: 550.163.01     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4090        Off |   00000000:01:00.0  On |                  Off |
| 30%   41C    P8             28W /  450W |   15925MiB /  24564MiB |     40%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A     13009      G   /usr/lib/xorg/Xorg                            240MiB |
|    0   N/A  N/A     13327      G   /usr/bin/gnome-shell                          106MiB |
|    0   N/A  N/A   2005144      G   /usr/share/code/code                           78MiB |
|    0   N/A  N/A   3208466      G   obs                                            70MiB |
|    0   N/A  N/A   3343950      G   ...seed-version=20250807-180039.874000         68MiB |
|    0   N/A  N/A   3602292      G   ...an,WebOTP --variations-seed-version         19MiB |
+-----------------------------------------------------------------------------------------+

ZhengyiLuo, 2025-08-21:
Is there a error message? Could be OOM on CPU or GPU.

onlyloveyanzi, 2025-08-22:
After setting the max_Len of motionlib to below 1000, this problem no longer occurs

closed 2025-08-22 by onlyloveyanzi